llvm/circt #402: [FIRRTL] Comparison Verilator Lint warning

drom, 2021-01-05:
This not a BUG

As part of an effort to make emitted Verilog Lint clean.

The following FIRRTL programs,
compiled with `firtool --lower-to-rtl` produces Verilog,
`verilator --lint-only` reports the following Warning:

True for all comparison operators.

## %Warning-CMPCONST : Comparison is constant due to limited range

```python
circuit top_mod :
  module top_mod :
    output tmp28: UInt<1>
    wire tmp11: UInt<1>
    tmp11 <= UInt<1>(1)
    tmp28 <= lt(tmp11, UInt<1>(1))
```

```verilog
module top_mod(
  output tmp28);

  wire tmp11;	// a_top_mod.fir:4:5

  assign tmp11 = 1'h1;	// a_top_mod.fir:5:{11,14}
  assign tmp28 = tmp11 < 1'h1;	// a_top_mod.fir:2:3, :5:14, :6:14
endmodule
```

```verilog
%Warning-CMPCONST: a_top_mod_new.v:7:24: Comparison is constant due to limited range
                                       : ... In instance top_mod
    7 |   assign tmp28 = tmp11 < 1'h1;  
      |                        ^
                   ... Use "/* verilator lint_off CMPCONST */" and lint_on around source to disable this message.
```

## %Warning-UNSIGNED : Comparison is constant due to unsigned arithmetic

```python
circuit top_mod :
  module top_mod :
    input inp_0: SInt<11>
    output tmp12: UInt<1>
    output tmp2: UInt<1>
    wire tmp6: UInt<28>
    tmp2 <= geq(SInt<13>(0), inp_0)
    tmp6 <= mul(UInt<27>(1), tmp2)
    tmp12 <= gt(UInt<1>(0), tmp6)
```

```verilog
module top_mod(
  input  [10:0] inp_0,
  output        tmp12, tmp2);

  wire        _T;	// a_top_mod.fir:3:3
  wire [27:0] tmp6;	// a_top_mod.fir:7:5

  assign _T = $signed(13'h0) >= $signed({{2{inp_0[10]}}, inp_0});	// a_top_mod.fir:8:{10,13,17}
  assign tmp6 = {{27'd0}, _T} * 28'h1;	// a_top_mod.fir:9:{10,13}
  assign tmp12 = 28'h0 > tmp6;	// a_top_mod.fir:3:3, :10:14
  assign tmp2 = _T;	// a_top_mod.fir:3:3, :10:14
endmodule
```

```verilog
%Warning-UNSIGNED: a_top_mod_new.v:10:24: Comparison is constant due to unsigned arithmetic
                                        : ... In instance top_mod
   10 |   assign tmp12 = 28'h0 > tmp6;  
      |                        ^
                   ... Use "/* verilator lint_off UNSIGNED */" and lint_on around source to disable this message.
```

lattner, 2021-01-10:
It looks like this is a request to strength reduce and fold comparisons.  For example

1) `a ugt 0` --> `a ne 0`
2) `a ugte 1` --> `a ugt 0` --> `a ne 0`
3) `a ult 0` --> `false`
4) `a ugt 0` --> true`

We should also canonicalize the constant onto the RHS:

5) `42 slt x` -> `x sgt 42`

lattner, 2021-01-10:
cc @youngar and @uenoku in case you're interested in this, it should be similar to other op canonicalization / folds.

youngar, 2021-01-22:
Some cases of strength reduction and constant folding were implemented for comparisons in #448.  This eliminated many lint warnings, but not all. The rest of the warnings can be solved with some form of constant propagation. We are still working on figuring out when wires should persist through optimization, and for now only eliminate wires which are never read from. There is some additional information here: https://github.com/llvm/circt/pull/456#pullrequestreview-568879579
```
// folded:
tmp32 <= _T < _T_4;
// not folded
assign tmp25 = tmp19 < tmp15;
```

lattner, 2021-01-24:
Hey @youngar I don't understand your comment immediately above.  How can those be folded?  What warning is verilator producing?

youngar, 2021-01-25:
Hey @lattner, I updated my example to make more sense.  I was just saying that although the canonicalization for ICmp was added, we are not optimizing around wires enough for all instances of the original lint warning (generated by the fuzzer) to disappear.

lattner, 2021-01-30:
Ok, thanks.  I think we should close this bug and discuss elimination of wires as a separate issue.  it is easy enough to remove all of them, but it isn't obvious this is desirable :-).  We need to figure out the "don't touch" story for the RTL dialect.

lattner, 2021-03-14:
Comparison folding is done.

closed 2021-03-14 by lattner