llvm/circt #407: [ExportVerilog] Instance name collisions

teqdruid, 2021-01-06:
https://github.com/llvm/circt/blob/27155e4d1740dc44bd596a0a04ca961b2136d014/lib/Translation/ExportVerilog/ExportVerilog.cpp#L1494

ExportVerilog just spits out instance names. If you've got two instances with the same name (as I'd imagine is frequent in generated code), they will conflict in the output. Those names may also conflict with other identifiers, such as wires and ports.

We should at least error out in the case where an instanceName conflicts with another idenifier. I think we should go further and automatically rename the instance.

lattner, 2021-01-08:
I agree with autorename.  I think that each module should have an instance name table similar to `nameTable` but for instances.

mikeurbach, 2021-01-18:
Is anyone working on this? I can take a crack at it.

teqdruid, 2021-01-18:
I'm not and it's a good bet nobody else is either given that its been here for awhile.

closed 2021-01-22 by mikeurbach