llvm/circt #423: [FIRRTL] Relax or Promote Connect to Partial Connect Semantics

seldridge, 2021-01-09:
Chisel will currently emits connects between a wider source and a narrower sink. The Scala FIRRTL Compiler accepts these and truncates the source (which is incorrect connect semantics, but correct partial connect semantics). This results in circuits that come out of Chisel which CIRCT won't accept.

Either:

1. We relax the connection verification checks 
2. We promote connects to partial connects if we see this during parsing

Example circuit:

```
circuit Foo :
  module Foo :
    input a: UInt<2>
    output b: UInt<1>

    b <= a
```

The SFC produces:

```verilog
module Foo(
  input  [1:0] a,
  output       b
);
  assign b = a[0];
endmodule
```

CIRCT gives us:

```bash
firtool Foo.fir
# Foo.fir:6:7: error: destination width 1 is not greater than or equal to source width 2
#    b <= a
#      ^
# Foo.fir:6:7: note: see current operation: "firrtl.connect"(%b, %a) : (!firrtl.flip<uint<1>>, !firrtl.uint<2>) -> ()
```

lattner, 2021-01-10:
I'd vote for #2

darthscsi, 2021-01-14:
So I accidentally went with 3. insert truncation during parsing.
Partial connect operation might be a better way to handle this.

darthscsi, 2021-01-18:
I've implemented #2.

PR #458 (2021-01-22): [FIRRTL] Deal with truncations before connects
Fix #428 
Fix #423
closed 2021-01-22 by darthscsi