llvm/circt #426: [FIRRTL] MLIR parser crashes on zero bit constants

lattner, 2021-01-10:
This in an mlir file crashes the mlir parser:
```
     %c0_ui0 = firrtl.constant(0 : ui0) : !firrtl.uint<0>
```

@seldridge are zero bit constants supported by SFC?  If so, I'll fix this to be allowed.  If not, I'll fix them to be rejected.

seldridge, 2021-01-10:
The SFC will reject all zero width constants citing that their width is too small.

This is a specific case of a constant being too large for the declared bit width. E.g., you get the same error if you try to parse a `UInt<1>(2)`. 

Note: it looks like CIRCT is parsing `UInt<1>(2)` into `%c0_ui1 = firrtl.constant(0 : ui1) : !firrtl.uint<1>`. I expect this can be solved generally.

Example:

```
circuit Foo:
  module Foo:
    output a: UInt<0>
    a <= UInt<0>(0)
```

```bash
firrtl -i Foo.fir   
# Exception in thread "main" firrtl.passes.CheckWidths$WidthTooSmall:  : [target ~Foo|Foo]  Width too small for constant 0.
```

Example:

```
circuit Foo:
  module Foo:
    output a: UInt<1>
    a <= UInt<1>(2)
```

```mlir
module  {
  firrtl.circuit "Foo" {
    firrtl.module @Foo(%a: !firrtl.flip<uint<1>>) {
      %c0_ui1 = firrtl.constant(0 : ui1) : !firrtl.uint<1>
      firrtl.connect %a, %c0_ui1 : !firrtl.flip<uint<1>>, !firrtl.uint<1>
    }
  }
}
```

lattner, 2021-01-11:
Fixed in ada31f1.  Please let me know if this starts rejecting anything we should keep.  It isn't clear if the signed restrictions apply to the hex and other forms.

closed 2021-01-11 by lattner

lattner, 2021-01-11:
Incidentally, hte mlir parser crash fix is [posted here](https://reviews.llvm.org/D94384)