llvm/circt #428: [FIRRTL] (dshl) error

drom, 2021-01-12:
The following FIRRTL program

```python
circuit top_mod :
  module top_mod :
    input inp_1: UInt<1>
    input inp_2: SInt<5>
    output tmp9: SInt<3>
    tmp9 <= dshl(inp_2, inp_1)
```

Compiled with `firtool --lower-to-rtl` gives an error:

```mlir
a_top_mod.fir:6:10: error: destination width 3 is not greater than or equal to source width 6
    tmp9 <= dshl(inp_2, inp_1)
         ^
a_top_mod.fir:6:10: note: see current operation: "firrtl.connect"(%tmp9, %0) : (!firrtl.flip<sint<3>>, !firrtl.sint<6>) -> ()
```

Compiled with `firrtl-1.4.0` produces this Verilog:

```verilog
module top_mod(
  input        inp_1,
  input  [4:0] inp_2,
  output [2:0] tmp9
);
  wire [5:0] _GEN_0 = {{1{inp_2[4]}},inp_2};
  wire [5:0] _GEN_1 = $signed(_GEN_0) << inp_1;
  assign tmp9 = _GEN_1[2:0];
endmodule
```

darthscsi, 2021-01-14:
According to the firrtl spec 7.14, this isn't a valid result type for dshl ( it should be 6).  The question is what to do with the implicit truncation.

seldridge, 2021-01-14:
I don't think there's anything going on with `dshl`, more that the SFC treats connect as having partial connect width truncation semantics: https://github.com/llvm/circt/issues/423.

seldridge, 2021-01-14:
Correction, there's something weird going on with `dshl` and possibly other ops where the SFC's `Legalize` pass is also adding casting. I'm trying to figure out what's going on.

seldridge, 2021-01-14:
Further correction: there's nothing wrong with the SFC here. The SFC blindly adds casts in this situation, but only because it already type checked `tmp <= dshl(...)`. I think this is then a duplicate of #423.

lattner, 2021-01-14:
Are we implementing dshl correctly?  It isn't in the firrtl spec so I took a guess at the result of the width and lowering.

seldridge, 2021-01-15:
`dshl` looks a-okay. And while that _is_ in the spec, I think you're thinking of the undocumented `dshlw`. That should only show up in "optimized low FIRRTL" and is an artifact of making the FIRRTL IR to Verilog mapping easier because FIRRTL and Verilog disagree about width behavior (life is very hard when you don't have multiple IRs... 😃).

lattner, 2021-01-15:
Yeah you're right I was thinking of dshlw.  Is it ok? :-)

PR #458 (2021-01-22): [FIRRTL] Deal with truncations before connects
Fix #428 
Fix #423
closed 2021-01-22 by darthscsi