llvm/circt #437: [FIRRTL] (bits) parameters 0,0 (core dump)

drom, 2021-01-12:
The following FIRRTL program

```python
circuit top_mod :
  module top_mod :
    output tmp19: UInt<1>
    tmp19 <= bits(SInt<1>(-1), 0, 0)
```

Compiled with `firtool --lower-to-rtl` gives an error:

```mlir
firtool: /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/APInt.cpp:859: llvm::APInt llvm::APInt::trunc(unsigned int) const: Assertion `width < BitWidth && "Invalid APInt Truncate request"' failed.
PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace.
Stack dump:
0.      Program arguments: ../../llvm/circt/build/bin/firtool a_top_mod.fir --lower-to-rtl --verilog -o=a_top_mod_new.v
 #0 0x00000000004c16e3 llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/Unix/Signals.inc:565:22
 #1 0x00000000004c179a PrintStackTraceSignalHandler(void*) /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/Unix/Signals.inc:632:1
 #2 0x00000000004bf7a0 llvm::sys::RunSignalHandlers() /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/Signals.cpp:71:20
 #3 0x00000000004c1136 SignalHandler(int) /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/Unix/Signals.inc:407:1
 #4 0x00007f927f19f1d0 __restore_rt (/lib64/libpthread.so.0+0x141d0)
 #5 0x00007f927ec5eae5 raise (/lib64/libc.so.6+0x3dae5)
 #6 0x00007f927ec47864 abort (/lib64/libc.so.6+0x26864)
 #7 0x00007f927ec47749 _nl_load_domain.cold (/lib64/libc.so.6+0x26749)
 #8 0x00007f927ec57016 (/lib64/libc.so.6+0x36016)
 #9 0x00000000004e2e0a llvm::APInt::trunc(unsigned int) const /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/APInt.cpp:860:3
#10 0x000000000079391d circt::firrtl::BitsPrimOp::fold(llvm::ArrayRef<mlir::Attribute>) /home/drom/work/github/llvm/circt/build/../lib/Dialect/FIRRTL/FIRRTLFolds.cpp:256:76
#11 0x0000000000786255 mlir::LogicalResult mlir::Op<circt::firrtl::BitsPrimOp, mlir::OpTrait::ZeroRegion, mlir::OpTrait::OneResult, mlir::OpTrait::OneTypedResult<circt::firrtl::UIntType>::Impl, mlir::OpTrait::ZeroSuccessor, mlir::OpTrait::OneOperand, mlir::MemoryEffectOpInterface::Trait>::foldSingleResultHook<circt::firrtl::BitsPrimOp>(mlir::Operation*, llvm::ArrayRef<mlir::Attribute>, llvm::SmallVectorImpl<mlir::OpFoldResult>&) /home/drom/work/github/llvm/circt/build/../llvm/mlir/include/mlir/IR/OpDefinition.h:1699:53
#12 0x00000000006ca63e mlir::AbstractOperation::foldHook(mlir::Operation*, llvm::ArrayRef<mlir::Attribute>, llvm::SmallVectorImpl<mlir::OpFoldResult>&) const /home/drom/work/github/llvm/circt/llvm/mlir/include/mlir/IR/OperationSupport.h:140:3
#13 0x00000000006c3da4 mlir::Operation::fold(llvm::ArrayRef<mlir::Attribute>, llvm::SmallVectorImpl<mlir::OpFoldResult>&) /home/drom/work/github/llvm/circt/llvm/mlir/lib/IR/Operation.cpp:581:30
#14 0x000000000088b0ab mlir::OperationFolder::tryToFold(mlir::OpBuilder&, mlir::Operation*, llvm::SmallVectorImpl<mlir::Value>&, llvm::function_ref<void (mlir::Operation*)>) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Transforms/Utils/FoldUtils.cpp:198:13
#15 0x000000000088aa0a mlir::OperationFolder::tryToFold(mlir::Operation*, llvm::function_ref<void (mlir::Operation*)>, llvm::function_ref<void (mlir::Operation*)>, bool*) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Transforms/Utils/FoldUtils.cpp:101:13
#16 0x000000000086ef87 (anonymous namespace)::GreedyPatternRewriteDriver::simplify(llvm::MutableArrayRef<mlir::Region>, int) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Transforms/Utils/GreedyPatternRewriteDriver.cpp:191:21
#17 0x000000000086f2b2 mlir::applyPatternsAndFoldGreedily(llvm::MutableArrayRef<mlir::Region>, mlir::FrozenRewritePatternList const&, unsigned int) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Transforms/Utils/GreedyPatternRewriteDriver.cpp:258:35
#18 0x000000000086f19e mlir::applyPatternsAndFoldGreedily(llvm::MutableArrayRef<mlir::Region>, mlir::FrozenRewritePatternList const&) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Transforms/Utils/GreedyPatternRewriteDriver.cpp:238:1
#19 0x0000000000737b74 (anonymous namespace)::Canonicalizer::runOnOperation() /home/drom/work/github/llvm/circt/llvm/mlir/lib/Transforms/Canonicalizer.cpp:34:3
#20 0x0000000000959599 mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Pass/Pass.cpp:400:36
#21 0x00000000009597c5 mlir::detail::OpToOpPassAdaptor::runPipeline(llvm::iterator_range<llvm::pointee_iterator<std::unique_ptr<mlir::Pass, std::default_delete<mlir::Pass> >*, mlir::Pass> >, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Pass/Pass.cpp:441:15
#22 0x000000000095b6dc mlir::PassManager::run(mlir::Operation*) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Pass/Pass.cpp:854:74
#23 0x0000000000409601 processBuffer(std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer> >, llvm::raw_ostream&) /home/drom/work/github/llvm/circt/build/../tools/firtool/firtool.cpp:141:13
#24 0x0000000000409ac0 main /home/drom/work/github/llvm/circt/build/../tools/firtool/firtool.cpp:195:13
#25 0x00007f927ec49152 __libc_start_main (/lib64/libc.so.6+0x28152)
#26 0x00000000004090ee _start /home/abuild/rpmbuild/BUILD/glibc-2.32/csu/../sysdeps/x86_64/start.S:122:0
```

Compiled with `firrtl-1.4.0` produces this Verilog:

```verilog
module top_mod(
  output  tmp19
);
  assign tmp19 = 1'h1;
endmodule
```

drom, 2021-01-12:
A second `0` parameter causing a crash.

lattner, 2021-01-12:
Fixed in 39bd895, thanks!

closed 2021-01-12 by lattner