llvm/circt #438: [ExportVerilog] Wires are declared after they are referenced

teqdruid, 2021-01-12:
In the situation where an expression cannot be inlined and appears in the IR after its use, ExportVerilog will emit the use before the declaration. Generally speaking, this will likely occur when a cycle occurs.

Setting `emitInlineWireDecls` ([here](https://github.com/llvm/circt/blob/8cbe578d2363a9e97395d1b7af99d9ad3cac60bc/lib/Translation/ExportVerilog/ExportVerilog.cpp#L33)) to false mitigates the issue. Always putting the wire declarations at the beginning of the file is the simplest thing to do, but probably not what we want.

Should we set `emitInlineWireDecls` to false and treat optimal decl placement as a future cleanup/enhancement?

teqdruid, 2021-01-13:
Will post a test case tomorrow.

lattner, 2021-01-13:
Thx, I appreciate it.  I am not surprised that 'cyclic' graphs aren't handled properly yet but it should be straight-forward to fix.  Thanks!

teqdruid, 2021-01-14:
```
  rtl.module @shl(%a: i1) -> (i1 {rtl.name = "b"}) {
    %1 = rtl.add %0, %0 : i1
    %2 = rtl.sub %0, %1 : i1
    %0 = rtl.shl %a, %a : i1
    rtl.output %1 : i1
  }
```

emits

```
module shl(
  input  a,
  output b);

  wire _T = _T_0 + _T_0;        // /home/jodemme/circt/integration_test/EmitVerilog/lint.mlir:33:10
  wire _T_0 = a << a;   // /home/jodemme/circt/integration_test/EmitVerilog/lint.mlir:35:10
  assign b = _T;        // /home/jodemme/circt/integration_test/EmitVerilog/lint.mlir:36:5
endmodule
```

And while Verilator accepts it, Questa barfs:
```
-- Compiling module shl
** Error: lint.sv(49): (vlog-2730) Undefined variable: '_T_0'.
** Error (suppressible): lint.sv(50): (vlog-2388) '_T_0' already declared in this scope (shl) at lint.sv(49).
```

teqdruid, 2021-01-14:
I was able to repro it without a cycle.

lattner, 2021-02-14:
> I was able to repro it without a cycle.

Can you share a small testcase?  I assume this is just when the operations are out of order?

lattner, 2021-02-14:
Fixed in 20eefde

closed 2021-02-14 by lattner