llvm/circt #439: [ExportVerilog] Expressions in sub-blocks do not get emitted out-of-line appropriately

teqdruid, 2021-01-12:
If you have an expression which cannot be inlined in a sub-block (e.g. `AlwaysOp`), it will be inlined inappropriately. The problem is that [`collectNamesEmitDecls`](https://github.com/llvm/circt/blob/8cbe578d2363a9e97395d1b7af99d9ad3cac60bc/lib/Translation/ExportVerilog/ExportVerilog.cpp#L1656) is only run on the top level module.

I'm not sure the best way to fix this. We could run `collectNamesEmitDecls` recursively but that would declare wires unnecessarily at the module scope. We could run it in `ModuleEmitter::visitSV(<block ops>)` but that would use the module-level name collision detection, so it will pollute the module namespace. We could create a hierarchical name collision resolution, but I don't think that's worth the effort at this point.

teqdruid, 2021-01-13:
Will post a test case tomorrow.

lattner, 2021-01-13:
Cool, I'll definitely take a look at this when I get a chance, thank you!

teqdruid, 2021-01-14:
```
rtl.module @exprInlineTest(%clk: i1) {
  %c = rtl.constant (0:i32) : i32
  sv.always posedge %clk {
    %e = rtl.extract %c from 0 : (i32) -> i16
    sv.fwrite "%d" ( %e ) : i16
    sv.fwrite "%d" ( %e ) : i16
  }
}

// produces
module exprInlineTest(
  input clk);

  wire [31:0] _T = 32'h0;       // test/ExportVerilog/sv-dialect.mlir:146:8
  always @(posedge clk) begin   // test/ExportVerilog/sv-dialect.mlir:147:3
    $fwrite(32'h80000002, "%d", _T[15:0]);      // test/ExportVerilog/sv-dialect.mlir:148:10, :149:5
    $fwrite(32'h80000002, "%d", _T[15:0]);      // test/ExportVerilog/sv-dialect.mlir:148:10, :150:5
  end // always @(posedge)
endmodule
```

But without the `sv.always` block and it produces:

```
module exprInlineTest(
  input clk);

  wire [31:0] _T = 32'h0;       // test/ExportVerilog/sv-dialect.mlir:146:8
  wire [15:0] _T_0 = _T[15:0];  // test/ExportVerilog/sv-dialect.mlir:148:10
  $fwrite(32'h80000002, "%d", _T_0);    // test/ExportVerilog/sv-dialect.mlir:149:5
  $fwrite(32'h80000002, "%d", _T_0);    // test/ExportVerilog/sv-dialect.mlir:150:5
endmodule
```

teqdruid, 2021-01-14:
The above test case is valid SV either way. There are cases (like the upcoming bitcast #400 operator) where it would. Tried to put together a test case with the `rtl.extract` operation, but couldn't make that work as I expected.

teqdruid, 2021-01-14:
Here's one with the `rtl.extract` operator, which (by the current ExportVerilog logic) should never be inlined:

```
rtl.module @exprInlineTest(%clk: i1) {
  %c = rtl.constant (0:i32) : i32
  sv.always posedge %clk {
    %e = rtl.extract %c from 0 : (i32) -> i16
    %f = rtl.add %e, %e : i16
    sv.fwrite "%d" ( %f ) : i16
  }
}
```
```
module exprInlineTest(
  input clk);

  wire [31:0] _T = 32'h0;       // test/ExportVerilog/sv-dialect.mlir:146:8
  always @(posedge clk) begin   // test/ExportVerilog/sv-dialect.mlir:147:3
    $fwrite(32'h80000002, "%d", _T[15:0] + _T[15:0]);   // test/ExportVerilog/sv-dialect.mlir:148:10, :149:10, :150:5
  end // always @(posedge)
endmodule
```

lattner, 2021-01-31:
Question about this (if you know off hand).  Is it ok to declare wires in a begin/end block?  For example, would this be ok?

```
wire [31:0] _T = 32'h0;       // test/ExportVerilog/sv-dialect.mlir:146:8
  always @(posedge clk) begin   // test/ExportVerilog/sv-dialect.mlir:147:3
    wire [15:0] _T2 = _T[15:0];
    $fwrite(32'h80000002, "%d", _T2 + _T2;   // test/ExportVerilog/sv-dialect.mlir:148:10, :149:10, :150:5
  end // always @(posedge)
```
or do we need:

```
  wire [31:0] _T = 32'h0;       // test/ExportVerilog/sv-dialect.mlir:146:8
  wire [15:0] _T2;
  always @(posedge clk) begin   // test/ExportVerilog/sv-dialect.mlir:147:3
    attach _T2 = _T[15:0];
    $fwrite(32'h80000002, "%d", _T2 + _T2;   // test/ExportVerilog/sv-dialect.mlir:148:10, :149:10, :150:5
  end // always @(posedge)
```

?

teqdruid, 2021-01-31:
IIRC, it is not valid to declare `wire`s in an `always` block. It is allowed, however, to declare a `logic` in blocks. I think.

lattner, 2021-02-12:
TEstcase:

// CHECK-LABEL: exprInlineTestIssue439
rtl.module @exprInlineTestIssue439(%clk: i1) {
  %c = rtl.constant (0:i32) : i32
  sv.always posedge %clk {
    %e = rtl.extract %c from 0 : (i32) -> i16
    %f = rtl.add %e, %e : i16
    sv.fwrite "%d"(%f) : i16
  }
}

lattner, 2021-02-13:
Ok, 132dfb8 implements the basic support for this, but will lead to wires getting declared in always blocks.  Can you give it a try and let me know if the wires need to be pulled out to the top level?  Thanks!

teqdruid, 2021-02-13:
```
module exprInlineTestIssue439(
  input clk);

  wire [31:0] _T = 32'h0;	// /home/jodemme/circt/integration_test/EmitVerilog/lint.mlir:96:8
  always @(posedge clk) begin	// /home/jodemme/circt/integration_test/EmitVerilog/lint.mlir:98:3
    wire [15:0] _T_0 = _T[15:0];	// /home/jodemme/circt/integration_test/EmitVerilog/lint.mlir:99:10
    $fwrite(32'h80000002, "%d", _T_0 + _T_0);	// /home/jodemme/circt/integration_test/EmitVerilog/lint.mlir:100:10, :101:5
  end // always @(posedge)
endmodule
```

```
Command Output (stderr):
--
%Error: /home/jodemme/circt/build/integration_test/EmitVerilog/Output/lint.mlir.tmp1.sv:105:5: syntax error, unexpected wire
  105 |     wire [15:0] _T_0 = _T[15:0];
      |     ^~~~
%Error: Exiting due to 1 error(s)
```

Shoot and a miss -- not valid. When I replace `wire` with `logic`, however, it lints clean.

Thanks for looking at this.

lattner, 2021-02-14:
ok, so I've been reading a bit about this, both the spec and posts [like this one](https://stackoverflow.com/questions/33459048/what-is-the-difference-between-reg-and-wire-in-a-verilog-module) to understand the difference between logic and wire.  Verilog is wacky ;-)

I think that we can model these things as logic always, since they are single assignment.  I'll give it a try to see what happens.

teqdruid, 2021-02-14:
My understanding of 'logic' is that it can be used in place of wire or reg,
so I tend to use it instead of wire and reg. It also has the advantage of
being able to use it in structs and aggregate types, which is not true of
wire I think.

lattner, 2021-02-14:
yeah it turns out that wire and logic/reg are completely different things.  The former is a net declaration the later is a variable declaration.  They are confusingly similar in some ways but in the spec they are quite different.

teqdruid, 2021-02-14:
Is there a reason we don't just use 'logic' everywhere?

lattner, 2021-02-14:
Please give e2b106c a try

lattner, 2021-02-14:
Oh that wasn't enough, one more step needed.

lattner, 2021-02-14:
Ok, I think that e043814 should wrap this up, please let me know if not or if you see any other problems, thanks!

closed 2021-02-14 by lattner

teqdruid, 2021-02-15:
Confirmed. New integration test: 8cebdd197cf3891dacaafd7a0a47f9f72c96db14

teqdruid, 2021-02-15:
Well, shit. There is a difference between wire and logic. From 6.4:

```
wire w = vara & varb; // net with a continuous assignment
logic v = consta & constb; // variable with initialization
logic vw; // no initial assignment
assign vw = vara & varb; // continuous assignment to a variable
```

So we need `wire` when assigning an expression, but `logic` inside always blocks! (And -- of course -- this isn't caught by lint checks since it's valid verilog, but not what we want!)