llvm/circt #440: always_ff with reset condition/block

darthscsi, 2021-01-12:
Consider adding sv.always_ff block with optional reset condition.  always_ff is nice to use for register lowering, having resets tied to the op would allow easy uniqing.

darthscsi, 2021-01-14:
Also, tying the reset to the always_ff op makes it easier to keep one reset condition/style per always block, which some tools strongly prefer.
alwaysff (clock, (reset, sync|async)^) { normal region} { optional reset region }

drom, 2021-01-16:
`always_ff` is a SystemVerilog feature. It is convenient (for human coders) safeguard of old good `always @(posedge clock ... )` which is important to support for V97 V2001 emission.

darthscsi, 2021-01-18:
Yes, it's an sv feature, but it's an sv dialect.  That said, it lowers trivially to plain verilog.

PR #491 (2021-01-22): Always_ff support with explicit resets and reset style (sync/async)
Always merging is another patch, but one forthcoming.
Fixes #440
closed 2021-01-22 by darthscsi