llvm/circt #463: [FIRRTL][RTL][ExportVerilog] Instances with no names

mikeurbach, 2021-01-15:
I have a question regarding FIRRTL instances with no names, lowering to RTL, and exporting Verilog.

Here's a really stripped down example. All that really matters is I am starting with a FIRRTL instance that has no name:

```
module  {                                                                                                                                                                                                          
  firrtl.circuit "top" {                                                                                                                                                                                           
    firrtl.module @sub() {}                                                                                                                                                                                        
    firrtl.module @top() {                                                                                                                                                                                         
      %0 = firrtl.instance @sub : !firrtl.bundle<>                                                                                                                                                                 
    }                                                                                                                                                                                                              
  }                                                                                                                                                                                                                
}
```

The `name` attribute is marked optional, and the lowering pass from Handshake to FIRRTL makes use of this to generate IR similar to the above.

If I use FIRRTL to directly export verilog (`firtool -format=mlir -verilog`), I get the following output:

```
module sub();                                                                                                                                                                                                      
endmodule                                                                                                                                                                                                          
                                                                                                                                                                                                                   
module top();                                                                                                                                                                                                      
  sub _T (
  );                                                                                                                                                                                                               
endmodule 
```

Because of how instances are handled in `ExportVerilogFIRRTL`, the instance goes through `addName`/`getName` and a unique name is generated.

If I try to lower to RTL first (`firtool -format=mlir -lower-to-rtl -verilog`), I get a helpful error that explains `'rtl.instance' op requires attribute 'instanceName'`.

Should the LowerToRTL pass do something to generate names in this case? It's nice to be able to leave off the name for passes like Handshake that don't really care how the instances are identified.

teqdruid, 2021-01-15:
I don't think so. ExportVerilog should be generating and uniquing names: #407.

mikeurbach, 2021-01-15:
Yep if ExportVerilog handled #407 this could be resolved by specifying an empty string for names. That's actually what Handshake does right now, but because of #407 the effect is the same in the Verilog output.

I'm also wondering about the FIRRTL dialect and lowering to RTL though. If the name is optional in FIRRTL but required in RTL, something has to give. It seems like either FIRRTL should mark the name required, or do something to generate one when it's missing. Maybe there is another approach.

teqdruid, 2021-01-15:
I don't think the RTL dialect should require instance names. If it encounters an instance which doesn't have a name attribute it should generate one (presumably based on the module it's instantiating for readability).

If the FIRRTL dialect can do something smarter with the name -- in terms readability -- then it should do the initial naming and rely on ExportVerilog to make sure it doesn't conflict with any other identifiers in the same scope.

lattner, 2021-01-16:
I agree.  We should allow empty names and synthesize placeholders just like we do for wires.

mikeurbach, 2021-01-18:
That sounds good to me. If empty instance names are allowed and #407 is resolved, this is resolved.

mikeurbach, 2021-01-18:
Actually, we already allow empty names, so closing this.

closed 2021-01-18 by mikeurbach