llvm/circt #469: [ExportVerilog] Should emit unsigned/signed for IntegerTypes which are not signless

teqdruid, 2021-01-17:
While the RTL dialect has no notion of signs, this is not the only source of IntegerTypes for the SV dialect. If a dialect wants to print signed or unsigned integers (`wire signed [7:0] signed8BitInt`), there is currently no way to do that. I think it's fair to emit the SystemVerilog keywords `signed` and `unsigned` in module ports and wire/reg declarations for IntegerTypes which are not signless.

lattner, 2021-01-17:
Yes, I think it is very reasonable to do support signfull types in the RTL dialect (and verilog printer), and I think you filed a bug about this in structs which should be duped to here.

However, I don't think the *operators* should work on them.  This implies we will need conversions back and forth between signless and signfull types.

teqdruid, 2021-01-17:
Yes, it was #397. It's very related but not quite the same. (In large part because of your comment about an sv.struct.)

Are you saying that when we get around to rtl.struct ExportVerilog support, we should emit signs for embedded types? I think that's reasonable as well, given that the RTL dialect has an open type system.

lattner, 2021-01-18:
Yes, that's what I'm saying.  By virtue of allowing an open type system, we should support signed types in `rtl.struct`.  We don't really have an open type system in all ways, but that should be the goal.