llvm/circt #472: [ExportVerilog] `rtl.struct` support

teqdruid, 2021-01-18:
Add the ability to emit structs.

teqdruid, 2021-01-18:
@darthscsi Are you planning on doing this? I don't want to step on your toes if I find time to work on it.

darthscsi, 2021-01-18:
It's on my roadmap, but I haven't started.

teqdruid, 2021-01-18:
Excellent. Please post here if/when you start and I'll do the same.

darthscsi, 2021-01-25:
On my list this week.

teqdruid, 2021-01-25:
Excellent!

Since this is a non-trivial addition, I wanted to give some input on how it should be done. I'm mostly concerned with struct naming.

1) Output the necessary structs in verilog first. This requires names. There's two ways we could do names: a) Give `rtl.struct` an optional attribute: name.  b) Add to the SV dialect a `typedef` op which typedefs a name to a particular data type. I slightly prefer the latter option since 'names' are a pretty-print language-level thing. It would also allow a dev to assign multiple names to a type.
2) Refactor ExportVerilog to use a `printType` function rather than assuming that everything is a wire.
3) Use the new `printType` to print one of the names associated with the struct. Modify places where we need to add that type output (e.g. BitcastOp, ArrayCreateOp).

Each of these could be a PR.

What do you think?

teqdruid, 2021-01-26:
I bring up the PRs since I may have time this week to do part of this work.

darthscsi, 2021-01-26:
The PR is forth coming soon.  It won't have typedef support.  A separate PR that adds that op would be handy.

teqdruid, 2021-01-27:
Excellent! Thanks for doing this.

lattner, 2021-03-14:
Is this done?

teqdruid, 2021-03-14:
Yes

closed 2021-03-14 by teqdruid