llvm/circt #478: [FIRRTL] Aggregate Ports w/ LowerTypes crashes LowerToRTL

seldridge, 2021-01-18:
Putting any aggregate port in a FIRRTL memory and running `LowerTypes` followed by `LowerToRTL` causes a crash.

E.g., the following will crash with `firtool -enable-lower-types -lower-to-rtl`:

```
circuit Foo :
  module Foo :
    input clock: Clock
    input readAddr: UInt<4>
    input readEn: UInt<1>
    output readData: UInt<8>[2]

    mem memory:
      data-type => UInt<8>[2]
      depth => 16
      reader => r
      read-latency => 0
      write-latency => 1
      read-under-write => undefined

    memory.r.clk <= clock
    memory.r.en <= readEn
    memory.r.addr <= readAddr
    readData <= memory.r.data
```

```
Assertion failed: (entry && "bundle lowering was not set"), function getBundleLowering, file ../lib/Dialect/FIRRTL/Transforms/LowerTypes.cpp, line 400.
PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace.
Stack dump:
0.	Program arguments: firtool -lower-to-rtl -enable-lower-types -verilog Foo.fir
1.	2.	Stack dump without symbol names (ensure you have llvm-symbolizer in your PATH or set the environment var `LLVM_SYMBOLIZER_PATH` to point to it):
0  firtool                  0x000000010b92316d llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) + 61
1  firtool                  0x000000010b92371b PrintStackTraceSignalHandler(void*) + 27
2  firtool                  0x000000010b9213bb llvm::sys::RunSignalHandlers() + 123
3  firtool                  0x000000010b925a68 SignalHandler(int) + 232
4  libsystem_platform.dylib 0x00007fff203aad7d _sigtramp + 29
5  libsystem_platform.dylib 0x00000000000049b0 _sigtramp + 18446603339975466064
6  libsystem_c.dylib        0x00007fff202b9720 abort + 120
7  libsystem_c.dylib        0x00007fff202b89d6 err + 0
8  firtool                  0x000000010ba68ac3 (anonymous namespace)::FIRRTLTypesLowering::getBundleLowering(mlir::Value, llvm::StringRef) + 211
9  firtool                  0x000000010ba68810 (anonymous namespace)::FIRRTLTypesLowering::visitExpr(circt::firrtl::SubfieldOp) + 384
10 firtool                  0x000000010ba6867b void circt::firrtl::ExprVisitor<(anonymous namespace)::FIRRTLTypesLowering, void>::dispatchExprVisitor(mlir::Operation*)::'lambda'(auto)::operator()<circt::firrtl::SubfieldOp>(auto) const + 43
11 firtool                  0x000000010ba685c1 llvm::TypeSwitch<mlir::Operation*, void>& llvm::TypeSwitch<mlir::Operation*, void>::Case<circt::firrtl::SubfieldOp, circt::firrtl::ExprVisitor<(anonymous namespace)::FIRRTLTypesLowering, void>::dispatchExprVisitor(mlir::Operation*)::'lambda'(auto)&>(void&&) + 113
12 firtool                  0x000000010ba68445 llvm::TypeSwitch<mlir::Operation*, void>& llvm::detail::TypeSwitchBase<llvm::TypeSwitch<mlir::Operation*, void>, mlir::Operation*>::Case<circt::firrtl::SubfieldOp, circt::firrtl::SubindexOp, circt::firrtl::SubaccessOp, circt::firrtl::AddPrimOp, circt::firrtl::SubPrimOp, circt::firrtl::MulPrimOp, circt::firrtl::DivPrimOp, circt::firrtl::RemPrimOp, circt::firrtl::AndPrimOp, circt::firrtl::OrPrimOp, circt::firrtl::XorPrimOp, circt::firrtl::LEQPrimOp, circt::firrtl::LTPrimOp, circt::firrtl::GEQPrimOp, circt::firrtl::GTPrimOp, circt::firrtl::EQPrimOp, circt::firrtl::NEQPrimOp, circt::firrtl::CatPrimOp, circt::firrtl::DShlPrimOp, circt::firrtl::DShlwPrimOp, circt::firrtl::DShrPrimOp, circt::firrtl::ValidIfPrimOp, circt::firrtl::AsSIntPrimOp, circt::firrtl::AsUIntPrimOp, circt::firrtl::AsAsyncResetPrimOp, circt::firrtl::AsClockPrimOp, circt::firrtl::CvtPrimOp, circt::firrtl::NegPrimOp, circt::firrtl::NotPrimOp, circt::firrtl::AndRPrimOp, circt::firrtl::OrRPrimOp, circt::firrtl::XorRPrimOp, circt::firrtl::BitsPrimOp, circt::firrtl::HeadPrimOp, circt::firrtl::InvalidValuePrimOp, circt::firrtl::MuxPrimOp, circt::firrtl::PadPrimOp, circt::firrtl::ShlPrimOp, circt::firrtl::ShrPrimOp, circt::firrtl::TailPrimOp, circt::firrtl::AsPassivePrimOp, circt::firrtl::AsNonPassivePrimOp, circt::firrtl::StdIntCastOp, circt::firrtl::AnalogInOutCastOp, circt::firrtl::ExprVisitor<(anonymous namespace)::FIRRTLTypesLowering, void>::dispatchExprVisitor(mlir::Operation*)::'lambda'(auto)&>(circt::firrtl::ExprVisitor<(anonymous namespace)::FIRRTLTypesLowering, void>::dispatchExprVisitor(mlir::Operation*)::'lambda'(auto)&) + 37
13 firtool                  0x000000010ba682e1 llvm::TypeSwitch<mlir::Operation*, void>& llvm::detail::TypeSwitchBase<llvm::TypeSwitch<mlir::Operation*, void>, mlir::Operation*>::Case<circt::firrtl::ConstantOp, circt::firrtl::SubfieldOp, circt::firrtl::SubindexOp, circt::firrtl::SubaccessOp, circt::firrtl::AddPrimOp, circt::firrtl::SubPrimOp, circt::firrtl::MulPrimOp, circt::firrtl::DivPrimOp, circt::firrtl::RemPrimOp, circt::firrtl::AndPrimOp, circt::firrtl::OrPrimOp, circt::firrtl::XorPrimOp, circt::firrtl::LEQPrimOp, circt::firrtl::LTPrimOp, circt::firrtl::GEQPrimOp, circt::firrtl::GTPrimOp, circt::firrtl::EQPrimOp, circt::firrtl::NEQPrimOp, circt::firrtl::CatPrimOp, circt::firrtl::DShlPrimOp, circt::firrtl::DShlwPrimOp, circt::firrtl::DShrPrimOp, circt::firrtl::ValidIfPrimOp, circt::firrtl::AsSIntPrimOp, circt::firrtl::AsUIntPrimOp, circt::firrtl::AsAsyncResetPrimOp, circt::firrtl::AsClockPrimOp, circt::firrtl::CvtPrimOp, circt::firrtl::NegPrimOp, circt::firrtl::NotPrimOp, circt::firrtl::AndRPrimOp, circt::firrtl::OrRPrimOp, circt::firrtl::XorRPrimOp, circt::firrtl::BitsPrimOp, circt::firrtl::HeadPrimOp, circt::firrtl::InvalidValuePrimOp, circt::firrtl::MuxPrimOp, circt::firrtl::PadPrimOp, circt::firrtl::ShlPrimOp, circt::firrtl::ShrPrimOp, circt::firrtl::TailPrimOp, circt::firrtl::AsPassivePrimOp, circt::firrtl::AsNonPassivePrimOp, circt::firrtl::StdIntCastOp, circt::firrtl::AnalogInOutCastOp, circt::firrtl::ExprVisitor<(anonymous namespace)::FIRRTLTypesLowering, void>::dispatchExprVisitor(mlir::Operation*)::'lambda'(auto)>(circt::firrtl::ExprVisitor<(anonymous namespace)::FIRRTLTypesLowering, void>::dispatchExprVisitor(mlir::Operation*)::'lambda'(auto)&&) + 49
14 firtool                  0x000000010ba6825d circt::firrtl::ExprVisitor<(anonymous namespace)::FIRRTLTypesLowering, void>::dispatchExprVisitor(mlir::Operation*) + 61
15 firtool                  0x000000010ba64d60 circt::firrtl::FIRRTLVisitor<(anonymous namespace)::FIRRTLTypesLowering, void>::dispatchVisitor(mlir::Operation*) + 32
16 firtool                  0x000000010ba6319e (anonymous namespace)::FIRRTLTypesLowering::runOnOperation() + 670
17 firtool                  0x000000010c0f87e5 mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) + 661
18 firtool                  0x000000010c0f8ef7 mlir::detail::OpToOpPassAdaptor::runPipeline(llvm::iterator_range<llvm::pointee_iterator<std::__1::unique_ptr<mlir::Pass, std::__1::default_delete<mlir::Pass> >*, mlir::Pass> >, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) + 359
19 firtool                  0x000000010c110ccf mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_8::operator()(llvm::MutableArrayRef<mlir::OpPassManager>) const + 559
20 firtool                  0x000000010c10f002 mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_8 std::__1::for_each<llvm::SmallVector<mlir::OpPassManager, 1u>*, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_8>(llvm::SmallVector<mlir::OpPassManager, 1u>*, llvm::SmallVector<mlir::OpPassManager, 1u>*, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_8) + 82
21 firtool                  0x000000010c10ef73 void llvm::parallel::detail::parallel_for_each<llvm::SmallVector<mlir::OpPassManager, 1u>*, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_8>(llvm::SmallVector<mlir::OpPassManager, 1u>*, llvm::SmallVector<mlir::OpPassManager, 1u>*, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_8) + 499
22 firtool                  0x000000010c0fa734 void llvm::parallelForEach<llvm::SmallVector<mlir::OpPassManager, 1u>*, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_8>(llvm::SmallVector<mlir::OpPassManager, 1u>*, llvm::SmallVector<mlir::OpPassManager, 1u>*, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_8) + 180
23 firtool                  0x000000010c0f9d3f mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool) + 1279
24 firtool                  0x000000010c0f8b2d mlir::detail::OpToOpPassAdaptor::runOnOperation(bool) + 77
[... cut ...] ager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) + 359
35 firtool                  0x000000010c0fb6c1 mlir::PassManager::run(mlir::Operation*) + 657
36 firtool                  0x000000010b797356 processBuffer(std::__1::unique_ptr<llvm::MemoryBuffer, std::__1::default_delete<llvm::MemoryBuffer> >, llvm::raw_ostream&) + 1238
37 firtool                  0x000000010b796b01 main + 897
38 libdyld.dylib            0x00007fff20381621 start + 1
```

mikeurbach, 2021-01-18:
That code may have never supported memories...

https://github.com/llvm/circt/blob/e3fce71b14e01c0ac4e001b6c54c5bada7b9ed4c/lib/Dialect/FIRRTL/Transforms/LowerTypes.cpp#L255-L265

I see this issue was already opened: https://github.com/llvm/circt/issues/479. If that is implemented, I think that will define away the subfield ops that currently fail this assertion for the case of non-aggregate types.

To handle storing aggregate types in memories, I think we would need to add a visitor to LowerTypes for memory ops that works just like the one for instances, which stores a mapping that the failing code here can pick up.

seldridge, 2021-01-18:
Thanks for the info, @mikeurbach! I'm just trying to get a handle on exactly what we currently support.

I hadn't thought about #479 as a solution to this, but I'll investigate in that direction. 

For aggregate types in memories, the SFC handles this in one of two different ways: 
1. Memories are "lowered" where you get one memory per element
2. Memories are "simplified" into one ground type and the original aggregate structure is overlayed on that (e.g., `{a: UInt<8>, b: UInt<8>}` becomes `data: UInt<16>` and references to `a` are replaced with a bit select on `data`).

(2) is much saner and is probably what the type lowering here should be doing.

This memory stuff is on my agenda this week, so I'll try to get to all of this.

seldridge, 2021-02-03:
Closed via #538.

closed 2021-02-03 by seldridge