llvm/circt #480: [FIRRTL] Handle read-write memory ports

seldridge, 2021-01-18:
Currently, any read-write memory ports get dropped during conversion to RTL with an error (but no hard exit from `firtool`). These, as the error message suggests, should be split up into separate read and write ports sometime before conversion to RTL to align with how the SFC handles these. Alternatively, these could be directly lowered to some structure that more directly expressions a true read/write port.

The main question is whether this should be a new pass or if this should be handled during an existing pass/conversion.

### Example

Consider the following FIRRTL circuit:

```
circuit Foo :
  module Foo :
    input clock: Clock
    input rwMode: UInt<1>
    input rwEn: UInt<1>
    input rwAddr: UInt<1>
    input rwMask: UInt<1>
    input rwDataIn: UInt<8>
    output rwDataOut: UInt<8>

    mem memory:
      data-type => UInt<8>
      depth => 16
      readwriter => rw
      read-latency => 0
      write-latency => 1
      read-under-write => undefined

    memory.rw.clk <= clock
    memory.rw.wmode <= rwMode
    memory.rw.en <= rwEn
    memory.rw.addr <= rwAddr
    memory.rw.wdata <= rwDataIn
    memory.rw.wmask <= rwMask
    rwDataOut <= memory.rw.rdata
```

The SFC will, during the `VerilogMemDelays` pass, split this into separate read and write ports. After that, the circuit looks like:

```
circuit Foo :
  module Foo :
    input clock : Clock
    input rwMode : UInt<1>
    input rwEn : UInt<1>
    input rwAddr : UInt<1>
    input rwMask : UInt<1>
    input rwDataIn : UInt<8>
    output rwDataOut : UInt<8>

    mem memory :
      data-type => UInt<8>
      depth => 16
      read-latency => 0
      write-latency => 1
      reader => rw_r
      writer => rw_w
      read-under-write => undefined
    rwDataOut <= memory.rw_r.data
    skip
    skip
    skip
    skip
    skip
    skip
    memory.rw_r.en <= and(rwEn, not(rwMode))
    memory.rw_r.addr <= pad(rwAddr, 4)
    memory.rw_r.clk <= clock
    memory.rw_w.en <= and(rwEn, rwMode)
    memory.rw_w.addr <= pad(rwAddr, 4)
    memory.rw_w.mask <= rwMask
    memory.rw_w.data <= rwDataIn
    memory.rw_w.clk <= clock
```

Currently, if we run the original FIRRTL IR through `firtool`, the memory is just dropped.

lattner, 2021-01-24:
I think we're going to have a "big lowering pass" of some sort and this sort of thing should go into it.  I'd name the pass LowerFIRRTL, and it would be a "firrtl dialect" to "firrtl dialect" pass, but with a bunch of stuff removed, including this and other stuff that is just annoying to handle in LowerToRTL.  We may eventually pull when lowering into this depending on the structure as well, since there is a general post-order traversal of hte tree that everything fits into, and there isn't a great need to make them independent passes.

closed 2021-03-02 by seldridge

seldridge, 2021-03-02:
Implemented this as another addition to `LowerTypes`. Note: that I'm fine if this is one of the things split out into the hypothetical "big lowering pass" and "lower types" or if "lower types" continues to evolve into "big lowering pass" (which is the direction that #691 pushes towards).

PR #691 (2021-03-02): Lower ReadWrite Ports during LowerTypes
Fixes #480.

Add lowering of FIRRTL Dialect memories with readwrite ports during the LowerTypes pass. This is accomplished by splitting each readwrite port into one read and one write port. The lowering of aggregate memories into ground type memories proceeds unaffected.

### Example

Starting with the following readwrite memory:

```scala
circuit Foo:
  module Foo:
    input clock: Clock
    input rwEn: UInt<1>
    input rwMode: UInt<1>
    input rwAddr: UInt<4>
    input rwMask: UInt<1>
    input rwDataIn: UInt<8>
    output rwDataOut: UInt<8>

    mem [... cut]