llvm/circt #492: [FIRRTL] Write Only Memory Causes Type Error

seldridge, 2021-01-21:
If you try to create a memory that only has a writer, the type of the memory will be `FlipType` due to type canonicalization. However, this then fails the ODS requirement that a memory is a `BundleType`. I ran into this from a different direction when working on #479/#488, but I think the problem may persist there.

Consider:

```
circuit Bar :
  module Bar :
    input clock: Clock
    input wAddr: UInt<4>
    input wEn: UInt<1>
    input wMask: UInt<1>
    input wData: UInt<8>

    mem memory:
      data-type => UInt<8>
      depth => 16
      writer => w
      read-latency => 0
      write-latency => 1
      read-under-write => undefined

    memory.w.clk <= clock
    memory.w.en <= wEn
    memory.w.addr <= wAddr
    memory.w.mask <= wMask
    memory.w.data <= wData
```

This will fail with:

```bash
firtool Bar.fir
# Bar.fir:9:5: error: 'firrtl.mem' op result #0 must be BundleType, but got '!firrtl.flip<bundle<w: bundle<addr: uint<4>, en: uint<1>, clk: clock, data: uint<8>, mask: uint<1>>>>'
#     mem memory:
#     ^
# Bar.fir:9:5: note: see current operation: %memory = "firrtl.mem"() {depth = 16 : i64, name = "memory", readLatency = 0 : i32, ruw = 0 : i32, writeLatency = 1 : i32} : () -> !firrtl.flip<bundle<w: bundle<addr: uint<4>, en: uint<1>, clk: clock, data: uint<8>, mask: uint<1>>>>
```

The SFC will return an empty module:

```
module Bar(
  input        clock,
  input  [3:0] wAddr,
  input        wEn,
  input        wMask,
  input  [7:0] wData
);
endmodule
```

I admit that intentional description of a memory hole is a weird one...

lattner, 2021-01-31:
You already fixed this, I added a testcase in 3ad2564.  At some point we'll have a dead code elimination cleanup pass that will eliminate these.

closed 2021-01-31 by lattner

seldridge, 2021-01-31:
Whoops. Yeah, part of the #488 work had to change the return type from `BundleType` to `VariadicType<FIRRTLType>`. Thanks for adding the test case. :+1: