llvm/circt #497: [RTL] ArrayIndexOp should accept ArrayType in addition to inout arrays

teqdruid, 2021-01-22:
Should we overload this Op? I think yes.

teqdruid, 2021-01-22:
@lattner Do you agree? If so, I'll do a PR now.

lattner, 2021-01-22:
SGTM.  We could either have two different ops, or overload.  My intuition pushes me towards thinking that two different ops will work better (easier to integrated into ODS etc) but I don't have a preference either way at this point, and I'm good with either approach.  Thank you for tackling this!

We should do the analogous thing for struct fields too at some point.

lattner, 2021-01-22:
Actually let me argue with myself here - I may be bringing C idioms into the HW world incorrectly:  Given value semantic arrays in SSA (e.g. what RTL and SIL have), you can implement `A[i] = 42` with these two pieces of pseudo IR:

Ainout = 
Iinout = gep Ainout, i
store 42 -> Iinout

And the value semantic approach of:

Avalue = ...
Avalue' = arrayinsert 42, i A

The reason that you want to be able to support the first thing in a C like language or something like LLVM IR is if you want to expose the intermediate pointer, e.g. across function boundaries.  This is required in C, but is not required in RTL.

For example, any time you want to do:
foo(&a[i])

You can instead do:

```
elt = a[i]
tmpWire = wire Int
store elt -> tmp
foo(tmpWire)

tmp2 = load tmpWire
arrayinsert tmp2, i, A
```

The later is MUCH MUCH easier to reason about because you define away aliasing issues and other things.

---

Is there a reason that you want to support interior pointers like this?

lattner, 2021-01-22:
Above, I'm arguing that perhaps we shouldn't have *any* interior pointer operations.  This would mean that you can't connect to subelements of a wire, only to the wire as a whole.

teqdruid, 2021-01-22:
First: the op I'm looking for is more precisely called something like "ArrayGetIndexOp". It gets the value at array[i]. I think this is consistent with SSA value semantics; however, given that it boils down to an alias of those wires (in SystemVerilog), it is affected by writes to the register which is driving those wires. All "values" in the RTL dialect are like this. "Value" semantics are kind of funky like that in a dialect which is inherently does not provide for data copies in pure combinational logic.

I get kind of confused when you talk about "pointers". Getting slices and indexes creates wire _aliases_ (though nobody refers to them like that). They are kinda related to pointers, but without the arithmetic so alias analysis isn't a problem. Also, we do support array slicing so I don't see an argument for array slicing but not indexing if the concern is about getting an alias into an array.

If the intent for ArrayIndexOp is to get some sort of "pointer" out of an inout array, then we should create a separate op. I don't really understand the InOutType since you keep talking about pointers in relation to it -- I kind of see how it could be considered a pointer to a shared memory, though it's not memory so the pointer analogy isn't great.

---

My use case boils down to this: I've got an array of ints. I need to get a slice of each element to reduced the number of bits (downcast) then construct a new array. This is the kind of operation I'd refer to as "slicing and dicing", which is pretty much what ESI does -- slices, dices, and transports data without any computation.

I was planning on looping through the array, getting the bitvector at each index, and slicing it down to the right size. (Special case required for signed numbers, though that's not relevant to this discussion.) At the end of the loop, I will need to re-assemble into a array of smaller width integers from those values.

This is pretty much a 'map' operation (in functional terminology). If I had an op which applied another op to all the elements of an array, that would be ideal, not bloat the IR, and perhaps be easier to reason about. I'm not really arguing for this op.

teqdruid, 2021-01-23:
I just looked a the usage of `ArrayIndexOp` and it does not do what I thought is does based on the name and documentation. It is used to produce a **reference** to an element in an `inout<array<...>>` which can then be used with an `rtl.connect`! (That documentation really needs a clean up.) This is **not** what I was trying to accomplish and I agree it is **not** needed. I'll do a PR for the op I was looking for. I'm pretty sure we're on the same page about that one.

Why is it needed for inout arrays? Presumably so one can drive only part of that inout array...? Is that really a common use case? I have no idea.

closed 2021-01-23 by teqdruid

lattner, 2021-01-24:
What I'm arguing is that we should eliminate things like "ArrayIndexOp getting an inout reference to a subelement of an inout thing".  That's what I think is broken (in my thinking) and should be eradicated.

teqdruid, 2021-01-24:
I agree (assuming we're finally speaking on the same wavelength).

lattner, 2021-01-31:
Ok, I just clarified a bunch of this.  We now have rtl.array_get and sv.array_index_inout which clarifies their behavior and what level of the stack they live in.  It turns out that we DO need the ability to get inout subindexes because we need to connect to them, e.g.:  `x[42] <= stuff`.