llvm/circt #507: [FIRRTL] Lower Types Fails for Certain Nested Bundles

seldridge, 2021-01-25:
The `LowerTypes` pass has problems with nested aggregates. Consider the following example:

```scala
circuit Foo :
  module Foo :
    input a: { b: {c: UInt<1>}}
    output b: { b: {c: UInt<1>}}

    b <= a
```

```bash
firtool -lower-to-rtl -enable-lower-types -verilog Foo.fir
# Assertion failed: (entry && "bundle lowering was not set"), function getBundleLowering, file ../lib/Dialect/FIRRTL/Transforms/LowerTypes.cpp, line 530.
```

I believe this is trying to find a lowering for the non-leaf types, but such a lowering was never recorded.

mikeurbach, 2021-01-25:
Hmm thanks for the reproducer.

When I first worked on this pass, I added a test with nested aggregates, but now that I look, it only tests nested aggregate _inputs_.

https://github.com/llvm/circt/blob/068f07c58ffc3911ea239e31da0e7c72b52d9fbb/test/Dialect/FIRRTL/lower-types.mlir#L69-L70

@seldridge I will take a look into this right now and report back. I have a feeling this has to do with the ways bundles with flips get canonicalized. There's probably somewhere we assume a bundle, and it needs to also check for a flip around a bundle, or something like that.

seldridge, 2021-01-25:
No need to drop everything! I'm just trying to log things as I come across them. Thanks, though!

mikeurbach, 2021-01-25:
I wasn't in the middle of anything, and I think it's a simple fix. In the end, it wasn't about bundle/flip canonicalization, but rather some naive code that hadn't been exercised with nested aggregates like your example. I'm sending a PR including your test case and the fix here in a second. Basically, there was a place that `flattenBundleTypes` wasn't called, and I think it should have been.

seldridge, 2021-01-26:
Awesome. Fix looks great. Thank you!

closed 2021-01-26 by mikeurbach

mikeurbach, 2021-01-26:
No problem, that was an oversight on my part.