llvm/circt #523: [RTL] Rename the RTL dialect to 'comb'

teqdruid, 2021-01-27:
What should the casing be on `comb`? All lower or `Comb`? I prefer all lower to save my wrists from the shift-c combo, but whatever.

lattner, 2021-01-28:
comb for the dialect, and Comb for the filenames. "CombOps.cpp" etc.

teqdruid, 2021-02-10:
Leave the module (and types?) in the RTL dialect.

lattner, 2021-02-12:
Yep, makes sense.  We should move the combinational ops to the comb dialect then figure out what remains and where it should go.

teqdruid, 2021-02-16:
I've started this effort here: https://github.com/llvm/circt/tree/rtl-comb

It's _way_ harder to extract only some ops.

teqdruid, 2021-02-16:
It's compiling now, but I'm getting `LLVM ERROR: Building op `comb.constant` but it isn't registered in this MLIRContext` in many of the tests. I'm inserting `comb::CombDialect` [here](https://github.com/llvm/circt/blob/0f52a5a8f3d6a83644504090bc2440457aec2c7e/include/circt/InitAllDialects.h#L33). What else must I do?

youngar, 2021-02-16:
@teqdruid If its the Conversion tests (something is converting TO the comb dialect), it has to be listed as a dependent dialect in the [ODS Pass declaration](https://github.com/llvm/circt/blob/main/include/circt/Conversion/Passes.td#L43)

teqdruid, 2021-02-16:
It's not just the comb tests. The ones exhibiting this error are at a minimum:
```
  CIRCT :: Dialect/ESI/cosim.mlir
  CIRCT :: Dialect/FIRRTL/lower-types-errors.mlir
```

The others are namespace issues, which may or may not turn into this once I fix them.

teqdruid, 2021-02-16:
Basically, any tests which run code with a `builder.create<comb::*>()`.

teqdruid, 2021-02-16:
turns out those tests had `circt-translate` in them so I had to add the new comb dialect to its registration calls.

youngar, 2021-02-22:
When building with `BUILD_SHARED_LIBS`, I'm getting a link error:
```
Undefined symbols for architecture x86_64:
  "circt::rtl::getBitWidth(mlir::Type)", referenced from:
      circt::comb::BitcastOp::verify() in CombOps.cpp.o
      circt::comb::__mlir_ods_local_type_constraint_Comb3(mlir::Operation*, mlir::Type, llvm::StringRef, unsigned int) in CombOps.cpp.o
```

I think this means we have to link Comb against RTL, but this is a circular dependency.

Have we considered moving all the RTL types (and `getBitWidth()`) over to the Comb dialect?

teqdruid, 2021-02-22:
Chris and I have been having a lengthy discussion about where to put the 'type' ops over in #604. This is another good reason not to have inter-dialect dependencies. It's also a good reason to move `bitcast` over to the rtl dialect.

I'm not aware of any logical deps from rtl to comb. I'll try removing that from cmake and see what goes boom.

teqdruid, 2021-02-22:
I forgot about the (near universal) dependency on `comb.constant`... It's only in one piece of code which I think only ESI uses.

teqdruid, 2021-02-22:
I fixed that circular dep, but there are others (mostly in ESI) and missing deps in exportverilog.

BTW, if you're building dynamic libs to speed up your builds, I've found that using lld rather then ld *significantly* reduces build time of static builds. `-DLLVM_ENABLE_LLD=ON`.

teqdruid, 2021-02-22:
See #641 for dynamic linking support fix

lattner, 2021-02-25:
Yes, @teqdruid is right about the universal dependency on comb.constant.  [I think it should be](https://github.com/llvm/circt/issues/665) hoisted back up to rtl/hw dialects.

teqdruid, 2021-02-25:
If we were to move the bitcast op back as well, I don't think there'd be any dependencies between rtl and comb.

lattner, 2021-02-26:
Where does bitcast get cross referenced?

teqdruid, 2021-02-26:
Bitcast uses `getBitWidth()` in the RTL dialect. I think that's the only reference between the two dialects in either direction at this point.

lattner, 2021-02-28:
Ok yeah, it is fine for comb/SV to call into RTL, just not the other way around.

closed 2021-03-14 by lattner