llvm/circt #524: [RTL/SV] Move all the inout stuff to the SV dialect

teqdruid, 2021-01-27:
This is an issue to discuss this topic.

teqdruid, 2021-01-27:
Arguments in favor:
 - It complicates the logic necessary in the RTL dialect. (See all the exceptions/look through logic.)
 - It (currently) has structure/syntactic semantics. The RTL dialect should be value semantics.
 - I get the feeling that this is a poorly understood concept in terms of generic/higher-level modeling. It is a well understood concept in SystemVerilog. There _may_ be a place for it in the RTL dialect once we get a handle on how it should be modeled.

lattner, 2021-01-27:
Yep, seems like the right thing to do to me.  inout type and wire should both move to SV.  This will also force us to figure out the type system layering issues.

lattner, 2021-01-31:
Fixed in 7ddfc3e.

closed 2021-01-31 by lattner

teqdruid, 2021-01-31:
Great! Thanks!