llvm/circt #526: Removing `using namespace mlir` from header files

youngar, 2021-01-27:
In several header files we have `using namespace mlir;`, which is considered bad C++ hygiene. Removing this from header files (and not source files) means explicitly qualifying each MLIR type with its namespace. This will affect all `*.td` and all `*.h` files in the include directory.

Alternatively, we could add declarations which selectively bridge the two namespaces, allowing us to use certain types unqualified.  We already import many common types from the LLVM namespace[0], and we can add MLIR types to this list.  Candidate types include `mlir::Operation`, `mlir::Value`, `mlir::MLIRContext` and all built-in attributes.

I don't have a preference for which way is best, and I don't mind doing the manual work either way.

[0] https://github.com/llvm/circt/blob/main/include/circt/Support/LLVM.h

teqdruid, 2021-01-27:
I have a slight preference to not importing anything into the public namespace -- I explicitly don't do that in the ESI dialect (I think). I do, however, use `using namespace` in my cpp files. But it's only a slight preference.

stephenneuendorffer, 2021-01-27:
This is mostly archaeology from a time when MLIR-tablgen didn't handle things in other namespaces very well.  I agree that it should be removed from header files.

youngar, 2021-01-27:
Thanks for the input, this is helpful information.  I think then we are leaning towards not importing any types, and eventually eliminating `circt/Support/LLVM.h` from CIRCT?

teqdruid, 2021-01-27:
> leaning towards not importing any types, and possibly eventually

Yes, I think so on both points.

youngar, 2021-01-28:
I opened a PR for this, and @lattner [expressed the opinion](https://github.com/llvm/circt/pull/529#issuecomment-769400851) that we should go the other way with this and selectively import types into the circt namespace.

@teqdruid if we do go the other way with this, shall I leave the ESI dialect to its own style, or remove the namespace qualifications where possible?

lattner, 2021-01-28:
Sorry I didn't see the comments above (I'm behind on email).  This isn't a tablegen specific issue.  This is about other C++ header files in the CIRCT world.  Adding llvm:: before arrayref everywhere and other standard types just obfucates the APIs.  This is why MLIR has its own LLVM.h and that pattern has worked well.

I agree that we shouldn't be "using namespace mlir/llvm;" in headers though!

teqdruid, 2021-01-28:
Your point about clutter is well taken. I prefer having the namespace on everything in the public API just so it's very clear from where things are coming but I don't care that much.  I didn't realize that MLIR did this -- I think we should replicate this convention in CIRCT.

@youngar If we're going to adopt a standard convention, it should be used everywhere for consistency. If you modify ESI to use that convention, I'd really appreciate it!

lattner, 2021-02-20:
Is this done?

youngar, 2021-02-20:
Fixed in https://github.com/llvm/circt/pull/586.

closed 2021-02-20 by youngar