llvm/circt #542: rtl.wire and sv.reg are weird

darthscsi, 2021-02-01:
rtl.wire and sv.reg serve the same purpose, to declare a named value.  However, they do so with different data types.  It's weird that
a) variable declarations get an op per type.  Reg and Wire are *types* in verilog.
b) There is no way to use any other primitive type in port or struct declarations.  We default to implicit wire, but reg and logic are also reasonable things to want.
c) There is no way to declare a variable of logic type.
d) standard types can map to multiple verilog types depending on the operation used.  Seems strange that type lowering is determined by operation in some contexts and not others.

darthscsi, 2021-02-01:
I would propose that rtl.wire and sv.reg are eliminated and replaced with sv.vardecl name : !sv.[logic|reg|wire]<stdinttype>.
Second, we need to declare the type that stdint types map to, so any stdint type not wrapped in a sv.[logic|reg|wire] will map to a consistent verilog type.

teqdruid, 2021-02-01:
I think we should default stdint to `logic` as it is the most flexible -- it can be used in code blocks and structs.

darthscsi, 2021-02-01:
Logic is nice in that it unifies the dealing of assignment types.  However, we also likely need to specify what we lower to for verilog.  And if we presume we need to output verilog also, then we will need to specify what happens here, even if it is to say that the result is context dependent (wire v.s. reg based on what we reduce assignments to, with the understanding that we will have to reduce all assignments of a variable to on form or another (procedural v.s. continuous)).

There is also the open question of wire v.s. bit, or more generally 4-state v.s. 2-state logic.  Which is brought up here.  Are stdint types 2-state?  If so, then bit would make more sense than logic.

lattner, 2021-02-05:
Yeah, I think it makes sense to merge these now that wire got sunk down to the SV dialect.  Adding an enum to specify how it gets printed makes sense to me.

lattner, 2021-02-14:
I dug into this a bit and I don't think it make sense to merge these.  These are actually two different things: wire is a specific case of a net declaration, and reg is a specific case of a variable declaration.  They have different constraints (e.g. you can't declare a net in an always block).  I think we should keep them separate.

lattner, 2021-02-20:
> I dug into this a bit and I don't think it make sense to merge these. These are actually two different things: wire is a specific case of a net declaration, and reg is a specific case of a variable [...]

Yes, I believe that even more so as I learn verilog.  I'm going to close this, but feel free to reopen if you disagree

closed 2021-02-20 by lattner