llvm/circt #558: [FIRRTL] Lower Vectors in LowerTypes

seldridge, 2021-02-04:
Currently, FIRRTL Dialect's lower types pass only supports bundles and not vectors. Add support for this.

Consider:

```scala
circuit Foo:
  module Foo:
    input a: UInt<1>[2]
    output b: UInt<1>[2]

    b <= a
```

Currently, CIRCT just won't lower these. With the Scala FIRRTL Compiler, these are blown out into:

```verilog
module Foo(
  input   a_0,
  input   a_1,
  output  b_0,
  output  b_1
);
  assign b_0 = a_0;
  assign b_1 = a_1;
endmodule
```

mikeurbach, 2021-02-06:
I never started on this, but when I was working on LowerTypes support for bundles, I thought this would be a natural extension. It should be relatively smooth to add to the flow in LowerTypes, with `subindex` handling similar to the existing `subfield` handling. I'm not sure how hard it would be to support `subaccess` as well, but that should work similarly.

seldridge, 2021-02-24:
@mikeurbach: You were spot on with the lowering being the same between `SubindexOp` and `SubfieldOp`. I'm going to split out `SubaccessOp` lowering into a separate issue as that's a bit of a different beast.

PR #655 (2021-02-26): [FIRRTL] Lower vectors, SubindexOp in LowerTypes
Add vector lowering and lowering of subindex in the LowerTypes pass.

Add one test that vectors are lowered correctly.

Fixes #558. 

This does not fix subaccess lowering. To do that in this PR would be substantially more invasive as that requires flattening the subaccess into a bunch of subaccess and a mux chain. I'll look at this in a follow-up PR.

I'm really unhappy about the amount of code copied between SubindexOp and SubfieldOp handling, but I was having some difficulty factoring this. I'd plan to clean this up in a later commit as long a [... cut]
closed 2021-02-26 by seldridge