llvm/circt #564: [FIRRTL] module instance is not fully initialized (false positive)

drom, 2021-02-06:
The following FIRRTL program

```python
circuit top_mod :
  module mod_0 :
    input tmp1: UInt<5>
    output tmp2: UInt<5>
    tmp2 <= tmp1
  module top_mod :
    input tmp1: UInt<5>
    output tmp2: UInt<5>
    inst U0 of mod_0
```

Compiled with `firtool --lower-to-rtl` produces legit Verilog, but:

Compiled with `firrtl-1.5.0` produces this error:

```python
Exception in thread "main" firrtl.passes.PassExceptions: 
firrtl.passes.CheckInitialization$RefNotInitializedException:  : [module top_mod]  Reference U0 is not fully initialized.
   : U0.tmp1 <= VOID
firrtl.passes.CheckInitialization$RefNotInitializedException:  : [module top_mod]  Reference tmp2 is not fully initialized.
   : tmp2 <= VOID
firrtl.passes.PassException: 2 errors detected!
```

lattner, 2021-02-15:
Right, we need to decide where and when to emit these errors.

drom, 2021-02-15:
I think firtool should report similar errors as firrtl.
One could initialize All ports with `U0 is invalid` command and then partially update connections one by one.

lattner, 2021-03-18:
Xref [this comment](https://github.com/llvm/circt/issues/779#issuecomment-801524871) on how we approach the design of this check

seldridge, 2021-03-18:
Note that these errors really suck for a couple of reasons:

1) The Scala FIRRTL Compiler is showing a FIRRTL IR snippet which the user has to mentally translate back to Chisel.
2) The "not fully initialized" isn't explained well to the user. What should they do to fix this? (Make a modification in Chisel, but the weak linkage of (1) makes this tedious... A Chisel user making a super basic error shouldn't have to know that a FIRRTL compiler exists.)

I think we can do way better here.

I also think this is in the "logical error" bucket that you identify in the cross reference.

Towards better error messages, I had started to sketch out ideas of what I wanted to see from [Chisel](https://github.com/chipsalliance/chisel3/issues/1435) and [FIRRTL](https://github.com/chipsalliance/firrtl/issues/1598). However, there was a huge infrastructural barrier to getting this working in Chisel and the Scala FIRRTL Compiler. Fortunately, MLIR/LLVM has that sorted. 😄

lattner, 2021-03-18:
👍

seldridge, 2021-03-23:
I changed the issue title to clarify that this is a general problem of checking initialization and not something specific to instances.

There are a number of tests that can be ported from the SFC: https://github.com/chipsalliance/firrtl/blob/master/src/test/scala/firrtlTests/CheckInitializationSpec.scala.

seldridge, 2021-08-31:
This should be fixed by https://github.com/llvm/circt/pull/885 or a follow-on commit/PR.

closed 2021-08-31 by seldridge