llvm/circt #569: error: no matching function for call to 'get'

oroppas, 2021-02-08:
I've got the following error during the build:

```
[17/120] Building CXX object tools/circt/lib/Dialect/FIRRTL/CMakeFiles/obj.CIRCTFIRRTL.dir/FIRRTLTypes.cpp.o
FAILED: tools/circt/lib/Dialect/FIRRTL/CMakeFiles/obj.CIRCTFIRRTL.dir/FIRRTLTypes.cpp.o
/usr/bin/clang++ -D_GNU_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -Itools/circt/lib/Dialect/FIRRTL -I/home/ryuta/packages/llvm/circt/lib/Dialect/FIRRTL -Iinclude -I/home/ryuta/packages/llvm/llvm-project/llvm/include -I/home/ryuta/packages/llvm/circt/include -Itools/circt/include -isystem /home/ryuta/packages/llvm/llvm-project/llvm/../mlir/include -isystem tools/mlir/include -stdlib=libc++ -fPIC -fvisibility-inlines-hidden -Werror=date-time -Werror=unguarded-availability-new -Wall -Wextra -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wmissing-field-initializers -pedantic -Wno-long-long -Wimplicit-fallthrough -Wcovered-switch-default -Wno-noexcept-type -Wnon-virtual-dtor -Wdelete-non-virtual-dtor -Wstring-conversion -fdiagnostics-color -ffunction-sections -fdata-sections -flto=thin -O3 -DNDEBUG  -fno-exceptions -std=c++14 -MD -MT tools/circt/lib/Dialect/FIRRTL/CMakeFiles/obj.CIRCTFIRRTL.dir/FIRRTLTypes.cpp.o -MF tools/circt/lib/Dialect/FIRRTL/CMakeFiles/obj.CIRCTFIRRTL.dir/FIRRTLTypes.cpp.o.d -o tools/circt/lib/Dialect/FIRRTL/CMakeFiles/obj.CIRCTFIRRTL.dir/FIRRTLTypes.cpp.o -c /home/ryuta/packages/llvm/circt/lib/Dialect/FIRRTL/FIRRTLTypes.cpp
In file included from /home/ryuta/packages/llvm/circt/lib/Dialect/FIRRTL/FIRRTLTypes.cpp:14:
In file included from /home/ryuta/packages/llvm/circt/include/circt/Dialect/FIRRTL/FIRRTLOps.h:62:
tools/circt/include/circt/Dialect/FIRRTL/FIRRTL.h.inc:2448:27: error: no matching function for call to 'get'
      return getPortNamed(StringAttr::get(name, getContext()));
                          ^~~~~~~~~~~~~~~
/home/ryuta/packages/llvm/llvm-project/llvm/../mlir/include/mlir/IR/BuiltinAttributes.h:331:21: note: candidate function not viable: no known conversion from 'mlir::MLIRContext *' to 'mlir::Type' for 2nd argument
  static StringAttr get(StringRef bytes, Type type);
                    ^
/home/ryuta/packages/llvm/llvm-project/llvm/../mlir/include/mlir/IR/BuiltinAttributes.h:328:21: note: candidate function not viable: no known conversion from 'llvm::StringRef' to 'mlir::MLIRContext *' for 1st argument
  static StringAttr get(MLIRContext *context, StringRef bytes);
                    ^
1 error generated.
```

Here's how I build

```
cmake \
  -G Ninja \
  -B llvm-build \
  -S llvm-project/llvm \
  -DBUILD_SHARED_LIBS:BOOL=ON \
  -DCMAKE_INSTALL_PREFIX:PATH=$HOME/local \
  -DCMAKE_INSTALL_RPATH:PATH=$HOME/local/lib \
  -DCMAKE_BUILD_WITH_INSTALL_RPATH:BOOL=ON \
  -DCMAKE_BUILD_TYPE:STRING=Release \
  -DLLVM_TARGETS_TO_BUILD:STRING=X86 \
  -DLLVM_ENABLE_PROJECTS:STRING="clang;clang-tools-extra;compiler-rt;lld;lldb;libcxx;libcxxabi;libunwind;mlir;flang;openmp" \
  -DLLVM_ENABLE_LIBCXX:BOOL=ON \
  -DLLVM_ENABLE_LLD:BOOL=ON \
  -DLLVM_ENABLE_LTO:STRING=Thin \
  -DLLVM_ENABLE_RTTI:BOOL=ON \
  -DLLVM_ENABLE_FFI:BOOL=OFF \
  -DLLVM_ENABLE_SPHINX:BOOL=OFF \
  -DLLVM_ENABLE_DOXYGEN:BOOL=OFF \
  -DLLVM_TOOL_GOLD_BUILD:BOOL=OFF \
  -DLLVM_INSTALL_UTILS:BOOL=ON \
  -DLLVM_BUILD_TESTS:BOOL=ON \
  -DLLVM_OPTIMIZED_TABLEGEN:BOOL=ON \
  -DLLVM_EXTERNAL_PROJECTS:STRING="circt;mlir-npcomp" \
  -DLLVM_EXTERNAL_CIRCT_SOURCE_DIR:PATH=$HOME/packages/llvm/circt \
  -DLLVM_EXTERNAL_MLIR_NPCOMP_SOURCE_DIR:PATH=$HOME/packages/llvm/mlir-npcomp \
  -DMLIR_BINDINGS_PYTHON_ENABLED:BOOL=ON \
  -DCLANG_DEFAULT_LINKER:STRING=lld \
  -DCLANG_DEFAULT_CXX_STDLIB:STRING=libc++ \
  -DCLANG_DEFAULT_RTLIB:STRING=compiler-rt \
  -DCLANG_DEFAULT_UNWINDLIB:STRING=libunwind \
  -DFLANG_BUILD_NEW_DRIVER:BOOL=ON \
  -DCOMPILER_RT_INCLUDE_TESTS:BOOL=ON \
  -DCOMPILER_RT_USE_BUILTINS_LIBRARY:BOOL=ON \
  -DLIBUNWIND_USE_COMPILER_RT:BOOL=ON \
  -DLIBCXXABI_USE_COMPILER_RT:BOOL=ON \
  -DLIBCXXABI_USE_LLVM_UNWINDER:BOOL=ON \
  -DLIBCXX_USE_COMPILER_RT:BOOL=ON \
  -DLIBCXX_CXX_ABI:STRING=libcxxabi \
  -DLIBCXXABI_USE_LLVM_UNWINDER:BOOL=ON \
  -DNPCOMP_ENABLE_IREE:BOOL=ON \
  -DNPCOMP_ENABLE_PYTORCH:STRING=OFF
```

mikeurbach, 2021-02-08:
I suspect this has to do with the version of LLVM you are building with. I see you are building both CIRCT and NPComp as external LLVM projects. Since both CIRCT and NPComp have their own expectation for the LLVM version, you will need to take care to pick a version that satisfies the expectations of both projects.

oroppas, 2021-02-09:
Thank you for the information. My apologies for misunderstanding of the LLVM requirement of CIRCT and NPComp.

closed 2021-02-09 by oroppas