llvm/circt #572: Assigning to constants!

darthscsi, 2021-02-08:
```
  ; CHECK-LABEL: module IsInvalid(
  ; CHECK-NEXT:    input a);
  module IsInvalid :
    output a: Analog<1>
    a is invalid  @[Place.scala 101]
```
Generates this awesome verilog:
```
module IsInvalid(
  input a);

  `ifndef SYNTHESIS
    alias a = '0;       // Place.scala:101
  `endif
  `ifdef SYNTHESIS
    assign '0 = a;      // Place.scala:101
    assign a = '0;      // Place.scala:101
  `endif // SYNTHESIS
endmodule
```

lattner, 2021-02-10:
How is "is invalid" lowered by SFC for analog types?

seldridge, 2021-02-11:
It looks like the SFC just drops it:

```verilog
module Foo(
  inout  [3:0] a
);
endmodule
```

The `RemovValidIf` transform is replacing the `is invalid` with a `skip`. Relevant code is [here](https://github.com/chipsalliance/firrtl/blob/cd845bdbfea0c09e9edbf61e651ede5197d8a084/src/main/scala/firrtl/passes/RemoveValidIf.scala#L46).

After that runs you have:

```
circuit Foo :
  module Foo :
    output a : Analog<4>
    skip
```

lattner, 2021-02-12:
Fixed in c883578, thanks!

closed 2021-02-12 by lattner