llvm/circt #578: Windows build break: missing symbol __imp_mlirContextRegisterRTLDialect

teqdruid, 2021-02-11:
```
ir.obj : error LNK2019: unresolved external symbol __imp_mlirContextRegisterRTLDialect referenced in function main [D:\a\circt\circt\build_release\test\CAPI\circt-capi-ir-test.vcxproj]
ir.obj : error LNK2019: unresolved external symbol __imp_mlirContextLoadRTLDialect referenced in function main [D:\a\circt\circt\build_release\test\CAPI\circt-capi-ir-test.vcxproj]
ir.obj : error LNK2019: unresolved external symbol __imp_mlirRTLDialectGetNamespace referenced in function main [D:\a\circt\circt\build_release\test\CAPI\circt-capi-ir-test.vcxproj]
D:\a\circt\circt\build_release\bin\circt-capi-ir-test.exe : fatal error LNK1120: 3 unresolved externals [D:\a\circt\circt\build_release\test\CAPI\circt-capi-ir-test.vcxproj]
```

Likely broken by b03db32b0f83824306c301b68ad02cb5a7af04b8. @GeorgeLyon Could you take a look please?

GeorgeLyon, 2021-02-11:
I think the patch updating MLIR will fix this, not sure how this sneaked past CI

GeorgeLyon, 2021-02-11:
Looks like @youngar fixed that patch, thanks Andrew! I've merged that PR and am re-running the windows build.

teqdruid, 2021-02-11:
the CI doesn't run the Windows build. It's run nightly. Not ideal, but we have a limited number of builders and the windows build is pretty slow, especially when compiling LLVM.

GeorgeLyon, 2021-02-11:
Looks like the build failed but with a different error: https://github.com/llvm/circt/runs/1881924297?check_suite_focus=true

teqdruid, 2021-02-11:
That's probably an under specification of the dependencies for the CAPI library or test.

GeorgeLyon, 2021-02-11:
The only dependency that should be needed for that symbol is this: https://github.com/llvm/circt/blob/162a0d824796c8423af0aa1dbf5e55257251ccd2/test/CAPI/CMakeLists.txt#L9
Indeed, that should be the only symbol in that target: https://github.com/llvm/circt/blob/162a0d824796c8423af0aa1dbf5e55257251ccd2/lib/CAPI/Dialect/RTL.cpp#L19

The "inconsistent visibility" message is curious too. I'm not sure how that is the case, though I am unfamiliar with Windows.
This is where it is declared:

https://github.com/llvm/llvm-project/blob/74916008a87dd2992d12af0b9725cb1804d0c8ba/mlir/include/mlir-c/Registration.h#L41
And this is where the definition is:

https://github.com/llvm/llvm-project/blob/2996a8d67553b9d469e01215b49bb1af17ad6d1e/mlir/include/mlir/CAPI/Registration.h#L48

GeorgeLyon, 2021-02-12:
@teqdruid Thinking about this a bit more, I'm wondering if the `MLIR_CAPI_EXPORTED` define isn't being set properly for some reason in one of the configurations. That would explain the "inconsistent visibility" message (one version has it correct and another has it incorrect), and incorrect visibility would explain why a symbol which seems to be pretty straightforwardly defined in a dependency, is not found. I'm not really sure how to test this theory in the Windows build.

teqdruid, 2021-02-12:
> The "inconsistent visibility" message is curious too. I'm not sure how that is the case, though I am unfamiliar with Windows

Where are you seeing this message?

I am also not terribly familiar with development on Windows.

GeorgeLyon, 2021-02-12:
Whoops, I misspoke, the log was not "inconsistent visibility" but "inconsistent dll linkage"
https://github.com/llvm/circt/runs/1881924297?check_suite_focus=true#step:7:171

teqdruid, 2021-02-12:
I see it now. I've had trouble with dynamic vs static linking on windows before with LLHD.

teqdruid, 2021-02-12:
I've repro'd the issue locally and established that if that function is exported properly with `__declspec(dllexport)` it links. (I hacked the header in llvm.) So you were correct about the capi exported being wrong.

teqdruid, 2021-02-12:
Ah... https://github.com/llvm/llvm-project/blob/4d3a061c32ee5c4fb88be389a4dd7fd71fc9fa9c/mlir/lib/CAPI/Dialect/CMakeLists.txt#L10 probably sets the macros properly.

teqdruid, 2021-02-12:
switching to `add_mlir_public_c_api_library` didn't fix the linker error :(

GeorgeLyon, 2021-02-12:
:( sorry this has become such an issue. If it ends up making sense to kick the can down the road, we can always put that macro directly into RTL.h (along with some lengthy prose) to guarantee it does the right thing.

GeorgeLyon, 2021-02-13:
OK, so `add_mlir_public_c_api_library` is on the right track, but it seems that some additional work happens here:
https://github.com/llvm/llvm-project/blob/5631842d181016207e85e4d035d0c4383f34337a/mlir/lib/CAPI/CMakeLists.txt#L11

If I'm understanding correctly, this will run _before_ the RTL dialect C API is added to MLIR_PUBLIC_C_API_LIBS (thanks, cmake). Maybe there is something we need to do in order to get those symbols into the shared library?

teqdruid, 2021-02-13:
It seems to me we have two handicaps: 1) nobody involved knows how cmake works and 2) nobody understands C++ development on Windows.

Let's just keep this issue open until someone has the time and willingness to look deeper into this rabbit hole.

teqdruid, 2021-07-28:
I just checked this test again and it works on Windows now! I suspect it was one of @mikeurbach's recent cleanup changes.

closed 2021-07-28 by teqdruid

mikeurbach, 2021-07-28:
Nice! The layers are slowly starting to align, so I guess that is going to appease the Windows gods.

teqdruid, 2021-07-28:
Yeah, so I'm starting to pay attention to windows.