llvm/circt #587: [RTL/SV] Pipeline generator

mikeurbach, 2021-02-12:
A common pattern seems to be generating an RTL (in the "register-transfer level") pipeline of some depth, most recently brought up here: https://github.com/llvm/circt/pull/585#issuecomment-778294457. This pattern appears in Handshake's [lowering](https://github.com/llvm/circt/blob/main/lib/Conversion/HandshakeToFIRRTL/HandshakeToFIRRTL.cpp#L1752-L1753) of memories for an analogous reason: to delay the control network to match the memory delay. Finally, ESI's buffers are [lowered](https://github.com/llvm/circt/blob/main/lib/Dialect/ESI/ESIPasses.cpp#L241) into pipeline stages implemented by an external module implemented in System Verilog.

It would be great if CIRCT could provide some sort of shared utility that these uses could all rely on.

teqdruid, 2021-02-12:
> (comment) <https://github.com/llvm/circt/pull/585#issuecomment-778294457>.

There are several very common patterns which are super easy to screw up.
I've been thinking a dialect which lowers into RTL/SV which contains these
higher-level constructs could be useful. E.g. Pipelines, FSMs, etc.

mikeurbach, 2021-02-12:
Putting those abstractions into a dialect sounds good to me. Now that I think about it, there is already a pipeline operation in the StaticLogic dialect: https://circt.llvm.org/docs/Dialects/StaticLogic/#staticlogicpipeline-circtstaticlogicpipelineop. I'm not sure what the intended layering of dialects is, but maybe we could standardize pipeline generators to go through that.

mikeurbach, 2021-04-13:
Coming back to this a couple months later, I guess the proper place for such an operation would be in the `seq` dialect? The `staticlogic.pipeline` is higher level, and solves a different problem.

teqdruid, 2021-04-13:
I don't think the `seq` dialect is the right place for this as it is a higher level construct and not well understood or agreed upon.

What would be the different between this op and the pipeline in static logic?

mikeurbach, 2021-04-13:
Maybe we're talking about different things. The StaticLogic pipeline is meant to describe pipelined computation: https://github.com/llvm/circt/blob/544f7473603e49101a29a61674d4efc492892e65/include/circt/Dialect/StaticLogic/StaticLogic.td#L35-L37

What I'm interested in is something more like the ESI ChannelBuffer: https://github.com/llvm/circt/blob/544f7473603e49101a29a61674d4efc492892e65/include/circt/Dialect/ESI/ESIOps.td#L26-L28

There are many places that we just want a bunch of "registers" connected in a series, without any fancy computation in between. So I was imagining something in the `seq` dialect (maybe "pipeline" is a misnomer), that essentially represents the same thing as the ChannelBuffer. It could potentially lower into `seq.compreg`s connected back-to-back. The ESI ChannelBuffer could trivially lower into this (or be replaced by it) and the FIRRTL memory lowering could also use it. Or, maybe this doesn't need to be a new op, it could just be a helper function to build a properly connected series of `seq.compreg`, similar to the code Andrew just implemented here: https://github.com/llvm/circt/pull/915/files#diff-d79ddd3f6d31d832cd0a2a7934495c126c0bad25a46fd09ca8b079a94c1fe815R60-R75

Apologies if I'm not understanding or if I'm using the wrong terminology, but the above pattern keeps coming up (so far in Handshake, ESI, and FIRRTL), so I thought it would be good to have one place in CIRCT where this can be implemented. Another way to look at it is I would like to generalize the ChannelBuffer to not be ESI-specific, and implement a lowering to in-tree dialects, rather than lowering to a Verilog primitive.

teqdruid, 2021-04-13:
Yeah, "pipeline" can mean many things but it _generally_ describes pipelined computation. There are a bunch of terms (all implying slightly different things) for what you're looking for. I typically use the term "buffered wire" (or just "buffer") but that is often confused with the notion of an *electrically* buffered wire. Pipelined wire is probably clearer.

I think there is value in an op like this. I actually agree that it should be in the `seq` dialect but I could imagine getting pushback on this. We don't have a talk scheduled for tomorrow. Maybe this could be a topic of discussion?

ESI's channel buffer will eventually broaden beyond what it is today, but I don't think I'll be able to lower it to a `PipelineRegOp` anytime soon since pipelines generally describe feed-forward (no backpressure) pipelines. ESI (and handshake) require _elastic_ pipelines, which are far more complex than a bunch of registers back-to-back. Any op which goes in `seq` would be feed-forward. FF pipelines can be used to implement elastic pipelines, but that's a who 'nother discussion.

mikeurbach, 2021-04-13:
Got it, thanks for the additional info. I'm looking forward to the memory discussion tomorrow, but I'll add this to the docket as well.

closed 2022-05-26 by teqdruid