llvm/circt #588: [HandshakeToFIRRTL] Buffers do not break cycles in the ready network.

mikeurbach, 2021-02-12:
While debugging #543, I found Verilator reported combinational cycles in the circuit. I've reviewed the BufferOp lowering, and it does indeed seem to only buffer the valid and ready signals. I have a couple thoughts about this:

This is another place we are generating pipeline register like #587. If there was a  solid lowering from staticlogic.pipeline to RTL/SV, we could lower to that here instead of FIRRTL.

We could borrow the [implementation](https://github.com/llvm/circt/blob/main/include/circt/Dialect/ESI/ESIPrimitives.sv) from ESI for inspiration, and eventually replace the primitive SV module there with a generated one.

For more inspiration, section 4.1 here discusses different approaches for data versus control buffers: http://www.cs.columbia.edu/~sedwards/papers/edwards2019compositional.pdf. The BufferOp in Handshake already has an attribute to indicate if it is a control buffer or not. We could update the BufferOp lowering to generate different circuits depending on the control attribute. Then in the `-handshake-insert-buffers` pass, I think it would be possible to insert a data and control buffer back-to-back like the paper describes.

I think for now the simplest thing would be to add to the current BufferOp lowering to also insert the appropriate FIRRTL to also buffer the ready network, similar to the paper and ESI. This can then be cleaned up as part of #587. Specializing the lowering based on the control attribute can be done afterwards if it is beneficial.

teqdruid, 2021-02-13:
I'm not sure the precise context here (and lack the mental energy to look into it at the moment), but there's nothing about ESI's PipelineStage which are ESI-specific -- it's a general valid/ready buffer. Would be a good candidate for this "grab bag" of common components dialect. Feel free to use it or move it.

The primitive it currently uses contains zero combinational paths from input token to output token -- it deals with the control and data buffering. It can stop on a dime at the cost of area. Anyway, while it is overkill in some situations it is typically functionally correct to use this module to buffer valid/ready tokens in a dynamically scheduled system.

I'd like to eventually replace the primitive with some lowering, but we don't have a `seq` dialect yet.

On another note, I can't find much on the StaticLogic dialect. It seems pretty minimal. Is anything using it?

mikeurbach, 2021-02-15:
> Is anything using it?

Nothing that I know of in the CIRCT tree.

> Feel free to use it or move it.
> I'd like to eventually replace the primitive with some lowering

I think it is possible to lower to the SV dialect (with some RTL sprinkled in?) to implement this today. This could go through the `seq` dialect instead, eventually.

I think there are two things I'd like to accomplish here:

1. Add lowerings for staticlogic.pipeline which use the implementation of PipelineStage from ESI. This could be done in a few incremental steps, but in the end state, I'd be shooting for a re-usable component that can be shared between Handshake and ESI (and potentially FIRRTL lowerings). I'd like for this component to lower to SV/RTL rather than an external module hand-implemented in Verilog. These enhancements are why I opened the issue #587.
2. Make it so HandshakeToFIRRTL doesn't lead to combinational cycles. This is more of a correctness issue than an enhancements, and it can be resolved independently of the above. I opened this issue to track the fact that I see cycles in the Handshake lowering right now, and I think we can close this issue when that is resolved, regardless of the implementation.

> I think for now the simplest thing would be to add to the current BufferOp lowering to also insert the appropriate FIRRTL to also buffer the ready network, similar to the paper and ESI. 

Over the weekend, I took a crack at this. It wasn't too bad to implement using a FIRRTL register and the appropriate logic, and it resolved the Verilator warnings about feedback loops. I am still debugging the generated circuit, but if this works for now, I'll make a PR.

mikeurbach, 2021-03-10:
Closed by https://github.com/llvm/circt/pull/601