llvm/circt #593: [FIRRTL] circuit with instance (core dump)

drom, 2021-02-15:
The following FIRRTL program

```python
circuit top_mod :
  module mod_2 :
    input clock: Clock
    input reset: UInt<1>
    input inp_a: {inp_d: UInt<14>}
  module top_mod :
    input clock: Clock
    input reset: UInt<1>
    inst U0 of mod_2
    U0 is invalid
    U0.reset <= reset
    U0.clock <= clock
```

Compiled with `firtool --lower-to-rtl --verilog` gives an error:

```mlir
firtool: ../lib/Dialect/FIRRTL/Transforms/LowerTypes.cpp:500: mlir::Value {anonymous}::FIRRTLTypesLowering::getBundleLowering(mlir::Value, llvm::StringRef): Assertion `entry && "bundle lowering was not set"' failed.
PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace.
Stack dump:
0.      Program arguments: ../../llvm/circt/build/bin/firtool a_top_mod.fir --lower-to-rtl --enable-lower-types --verilog -o=a_top_mod_new.v
1.      2.       #0 0x00000000004c54ed llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/Unix/Signals.inc:565:22
 #1 0x00000000004c55a4 PrintStackTraceSignalHandler(void*) /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/Unix/Signals.inc:632:1
 #2 0x00000000004c35a4 llvm::sys::RunSignalHandlers() /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/Signals.cpp:71:20
 #3 0x00000000004c4f40 SignalHandler(int) /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/Unix/Signals.inc:407:1
 #4 0x00007f6f180221d0 __restore_rt (/lib64/libpthread.so.0+0x141d0)
 #5 0x00007f6f17aa9ae5 raise (/lib64/libc.so.6+0x3dae5)
 #6 0x00007f6f17a92864 abort (/lib64/libc.so.6+0x26864)
 #7 0x00007f6f17a92749 _nl_load_domain.cold (/lib64/libc.so.6+0x26749)
 #8 0x00007f6f17aa2016 (/lib64/libc.so.6+0x36016)
 #9 0x00000000005b68be (anonymous namespace)::FIRRTLTypesLowering::getBundleLowering(mlir::Value, llvm::StringRef) /home/drom/work/github/llvm/circt/build/../lib/Dialect/FIRRTL/Transforms/LowerTypes.cpp:500:3
#10 0x00000000005b6a2f (anonymous namespace)::FIRRTLTypesLowering::getAllBundleLowerings(mlir::Value, llvm::SmallVectorImpl<mlir::Value>&) /home/drom/work/github/llvm/circt/build/../lib/Dialect/FIRRTL/Transforms/LowerTypes.cpp:521:22
#11 0x00000000005b62aa (anonymous namespace)::FIRRTLTypesLowering::visitStmt(circt::firrtl::ConnectOp) /home/drom/work/github/llvm/circt/build/../lib/Dialect/FIRRTL/Transforms/LowerTypes.cpp:429:3
#12 0x00000000005b84cf void circt::firrtl::StmtVisitor<(anonymous namespace)::FIRRTLTypesLowering, void>::dispatchStmtVisitor(mlir::Operation*)::'lambda'(auto)::operator()<circt::firrtl::ConnectOp>(auto) const /home/drom/work/github/llvm/circt/build/../include/circt/Dialect/FIRRTL/FIRRTLVisitors.h:161:13
#13 0x00000000005b82af llvm::TypeSwitch<mlir::Operation*, void>& llvm::TypeSwitch<mlir::Operation*, void>::Case<circt::firrtl::ConnectOp, circt::firrtl::StmtVisitor<(anonymous namespace)::FIRRTLTypesLowering, void>::dispatchStmtVisitor(mlir::Operation*)::'lambda'(auto)&>(void&&) /home/drom/work/github/llvm/circt/build/../llvm/llvm/include/llvm/ADT/TypeSwitch.h:159:18
#14 0x00000000005b8029 llvm::TypeSwitch<mlir::Operation*, void>& llvm::detail::TypeSwitchBase<llvm::TypeSwitch<mlir::Operation*, void>, mlir::Operation*>::Case<circt::firrtl::ConnectOp, circt::firrtl::DoneOp, circt::firrtl::MemoryPortOp, circt::firrtl::PartialConnectOp, circt::firrtl::PrintFOp, circt::firrtl::SkipOp, circt::firrtl::StopOp, circt::firrtl::WhenOp, circt::firrtl::AssertOp, circt::firrtl::AssumeOp, circt::firrtl::CoverOp, circt::firrtl::StmtVisitor<(anonymous namespace)::FIRRTLTypesLowering, void>::dispatchStmtVisitor(mlir::Operation*)::'lambda'(auto)&>(circt::firrtl::StmtVisitor<(anonymous namespace)::FIRRTLTypesLowering, void>::dispatchStmtVisitor(mlir::Operation*)::'lambda'(auto)&) /home/drom/work/github/llvm/circt/build/../llvm/llvm/include/llvm/ADT/TypeSwitch.h:41:42
#15 0x00000000005b7d8f llvm::TypeSwitch<mlir::Operation*, void>& llvm::detail::TypeSwitchBase<llvm::TypeSwitch<mlir::Operation*, void>, mlir::Operation*>::Case<circt::firrtl::AttachOp, circt::firrtl::ConnectOp, circt::firrtl::DoneOp, circt::firrtl::MemoryPortOp, circt::firrtl::PartialConnectOp, circt::firrtl::PrintFOp, circt::firrtl::SkipOp, circt::firrtl::StopOp, circt::firrtl::WhenOp, circt::firrtl::AssertOp, circt::firrtl::AssumeOp, circt::firrtl::CoverOp, circt::firrtl::StmtVisitor<(anonymous namespace)::FIRRTLTypesLowering, void>::dispatchStmtVisitor(mlir::Operation*)::'lambda'(auto)>(circt::firrtl::StmtVisitor<(anonymous namespace)::FIRRTLTypesLowering, void>::dispatchStmtVisitor(mlir::Operation*)::'lambda'(auto)&&) /home/drom/work/github/llvm/circt/build/../llvm/llvm/include/llvm/ADT/TypeSwitch.h:42:3
#16 0x00000000005b7bb2 circt::firrtl::StmtVisitor<(anonymous namespace)::FIRRTLTypesLowering, void>::dispatchStmtVisitor(mlir::Operation*) /home/drom/work/github/llvm/circt/build/../include/circt/Dialect/FIRRTL/FIRRTLVisitors.h:160:52
#17 0x00000000005b79a1 circt::firrtl::FIRRTLVisitor<(anonymous namespace)::FIRRTLTypesLowering, void>::visitInvalidExpr(mlir::Operation*) /home/drom/work/github/llvm/circt/build/../include/circt/Dialect/FIRRTL/FIRRTLVisitors.h:267:49
#18 0x00000000005b760c void circt::firrtl::ExprVisitor<(anonymous namespace)::FIRRTLTypesLowering, void>::dispatchExprVisitor(mlir::Operation*)::'lambda0'(auto)::operator()<mlir::Operation*>(auto) const /home/drom/work/github/llvm/circt/build/../include/circt/Dialect/FIRRTL/FIRRTLVisitors.h:54:56
#19 0x00000000005b7359 void llvm::TypeSwitch<mlir::Operation*, void>::Default<circt::firrtl::ExprVisitor<(anonymous namespace)::FIRRTLTypesLowering, void>::dispatchExprVisitor(mlir::Operation*)::'lambda0'(auto)>(auto&&) /home/drom/work/github/llvm/circt/build/../llvm/llvm/include/llvm/ADT/TypeSwitch.h:168:3
#20 0x00000000005b7066 circt::firrtl::ExprVisitor<(anonymous namespace)::FIRRTLTypesLowering, void>::dispatchExprVisitor(mlir::Operation*) /home/drom/work/github/llvm/circt/build/../include/circt/Dialect/FIRRTL/FIRRTLVisitors.h:55:10
#21 0x00000000005b6d23 circt::firrtl::FIRRTLVisitor<(anonymous namespace)::FIRRTLTypesLowering, void>::dispatchVisitor(mlir::Operation*) /home/drom/work/github/llvm/circt/build/../include/circt/Dialect/FIRRTL/FIRRTLVisitors.h:262:49
#22 0x00000000005b4992 (anonymous namespace)::FIRRTLTypesLowering::runOnOperation() /home/drom/work/github/llvm/circt/build/../lib/Dialect/FIRRTL/Transforms/LowerTypes.cpp:145:39
#23 0x0000000000988303 mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Pass/Pass.cpp:399:36
#24 0x000000000098852f mlir::detail::OpToOpPassAdaptor::runPipeline(llvm::iterator_range<llvm::pointee_iterator<std::unique_ptr<mlir::Pass, std::default_delete<mlir::Pass> >*, mlir::Pass> >, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Pass/Pass.cpp:440:15
#25 0x000000000098920a mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::'lambda'(llvm::MutableArrayRef<mlir::OpPassManager>)::operator()(llvm::MutableArrayRef<mlir::OpPassManager>) const /home/drom/work/github/llvm/circt/llvm/mlir/lib/Pass/Pass.cpp:616:68
#26 0x000000000098c371 mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::'lambda'(llvm::MutableArrayRef<mlir::OpPassManager>) std::for_each<llvm::SmallVector<mlir::OpPassManager, 1u>*, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::'lambda'(llvm::MutableArrayRef<mlir::OpPassManager>)>(llvm::SmallVector<mlir::OpPassManager, 1u>*, llvm::SmallVector<mlir::OpPassManager, 1u>*, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::'lambda'(llvm::MutableArrayRef<mlir::OpPassManager>)) /usr/include/c++/10/bits/stl_algo.h:3838:7
#27 0x000000000098c1c8 void llvm::parallel::detail::parallel_for_each<llvm::SmallVector<mlir::OpPassManager, 1u>*, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::'lambda'(llvm::MutableArrayRef<mlir::OpPassManager>)>(llvm::SmallVector<mlir::OpPassManager, 1u>*, llvm::SmallVector<mlir::OpPassManager, 1u>*, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::'lambda'(llvm::MutableArrayRef<mlir::OpPassManager>))::'lambda'()::operator()() const /home/drom/work/github/llvm/circt/llvm/llvm/include/llvm/Support/Parallel.h:140:38
[... cut ...]
#49 0x0000000000409afb main /home/drom/work/github/llvm/circt/build/../tools/firtool/firtool.cpp:197:13
#50 0x00007f6f17a94152 __libc_start_main (/lib64/libc.so.6+0x28152)
#51 0x00000000004090ee _start /home/abuild/rpmbuild/BUILD/glibc-2.32/csu/../sysdeps/x86_64/start.S:122:0
./process1.sh: line 24: 10266 Aborted                 (core dumped) ../../llvm/circt/build/bin/firtool a_top_mod.fir --lower-to-rtl --enable-lower-types --verilog -o=$VFILE2
```

Compiled with `firrtl-1.5.0` produces this Verilog:

```verilog
module top_mod(
  input   clock,
  input   reset
);
endmodule
```

lattner, 2021-02-21:
LowerTypes should handle this (which is tracked in another bug) but we shouldn't crash on unsupported stuff.  Fixed in 02e13d1, thanks!

drom, 2021-02-23:
The code above is still crashing `firtool`

PR #650 (2021-02-23): [FIRRTL] Handle invalidvalue of struct type in LowerTypes
Fixes #593
closed 2021-02-23 by darthscsi