llvm/circt #595: [ExportVerilog] Wire declared after use

teqdruid, 2021-02-16:
I've observed a wire declared after use:

```
module top(
  input [127:0] arr);

  assert(_T[6'h0+:32] == 32'h0);        // defuse3.mlir:2:13, :3:9, :4:3, :8:12, :9:9, :10:9
  logic [63:0] _T = arr[7'h0+:64];      // defuse3.mlir:6:12, :7:9
endmodule
```

Resulting from:
```
rtl.module @top(%arr: !rtl.array<128 x i1>) {
  %c0_i32 = rtl.constant(0 : i32) : i32
  %92 = rtl.icmp eq %19, %c0_i32 : i32
  sv.assert %92 : i1

  %c0_i7 = rtl.constant(0 : i7) : i7
  %17 = rtl.array_slice %arr at %c0_i7 : (!rtl.array<128xi1>) -> !rtl.array<64xi1>
  %c0_i6 = rtl.constant(0 : i6) : i6
  %18 = rtl.array_slice %17 at %c0_i6 : (!rtl.array<64xi1>) -> !rtl.array<32xi1>
  %19 = rtl.bitcast %18 : (!rtl.array<32xi1>) -> i32
}
```

teqdruid, 2021-02-16:
Same occurred when I replaced `logic` with `wire`.

teqdruid, 2021-02-16:
The verilog produced isn't valid since `assert` needs to be in an `always` block, but that's ancillary.

teqdruid, 2021-02-16:
Huh. Questa is now accepting declarations which come after use. I could've sworn it didn't like that. Nevermind.

lattner, 2021-02-16:
The verilog emitter shouldn't do this, I'll take a look.  Thank you for the testcase!

$ circt-translate x.mlir -export-verilog
```
module top(
  input [127:0] arr);

  assert(_T[6'h0+:32] == 32'h0);	// x.mlir:2:13, :3:9, :4:3, :8:12, :9:9, :10:9
  logic [63:0] _T = arr[7'h0+:64];	// x.mlir:6:12, :7:9
endmodule
```

lattner, 2021-02-22:
Fixed in 12808e5, thanks again

closed 2021-02-22 by lattner