llvm/circt #606: [FIRRTL] subindex op should supprt flips of vectors

youngar, 2021-02-17:
Running the FIRRTL compiler on benchmarks from SFC, the [HwachaSequencer.fir](https://github.com/chipsalliance/firrtl/blob/master/regress/HwachaSequencer.fir) fails with:
```
sequencer-master.scala:410:35: error: 'firrtl.subindex' op operand #0 must be FVectorType, but got '!firrtl.flip<vector<uint<1>, 8>>'
sequencer-master.scala:410:35: note: see current operation: %6643 = "firrtl.subindex"(%6642) {index = 0 : i32} : (!firrtl.flip<vector<uint<1>, 8>>) -> !firrtl.flip<uint<1>>
```

lattner, 2021-02-20:
Reduced testcase:

circuit HwachaSequencer :
  module HwachaSequencer : 
    output io : UInt<1>[8]
    io[0] <= UInt<1>("h00")

lattner, 2021-02-20:
Fixed in 49a3f99, thanks!

closed 2021-02-20 by lattner