llvm/circt #607: [FIRRTL] Modules should not enforce use-before-def

youngar, 2021-02-17:
Running the FIRRTL compiler on benchmarks from SFC, the [FPU.fir](https://github.com/chipsalliance/firrtl/blob/master/regress/FPU.fir) fails with:
```
// firtool --lower-to-rtl --enable-lower-types --pass-timing --verilog ./regress/FPU.fir 
./regress/FPU.fir:39:5: error: use of undefined module name 'FPUDecoder' in instance
    inst fp_decoder of FPUDecoder @[FPU.scala 520:26
```

seldridge, 2021-02-19:
As a temporary workaround you can re-order things using the SFC before passing them into the MFC.

```
firrtl -i Foo.fir -X high -fct firrtl.transforms.SortModules
```

This will technically lower to high FIRRTL from CHIRRTL, but will also run a transform that topologically sorts the modules so the MFC won't complain.

I'm torn on whether this is a FIRRTL spec bug or if we should allow use-before-def...

lattner, 2021-02-20:
Yes, this is a bug, tracked here:
https://github.com/llvm/circt/issues/163

closed 2021-02-20 by lattner