llvm/circt #610: [FIRRTL] `regress/Ops.fir` test failing

youngar, 2021-02-17:
The [regress/Ops.fir](https://github.com/chipsalliance/firrtl/blob/master/regress/Ops.fir) test is currently failing because we lack support for unknown width integers.  I figured I would report this so we make sure this test is working in the future.

```
./regress/Ops.fir:3:3: error: 'firrtl.pad' op result type should be '!firrtl.sint'
  module Ops:
  ^
./regress/Ops.fir:3:3: note: see current operation: %4 = "firrtl.pad"(%c0_si) {amount = 14 : i32} : (!firrtl.sint) -> !firrtl.sint<14>
```

A more minimal example:
```firrtl
circuit Ops:
  module Ops:
    output os: SInt<14>
    os <= SInt(0)
```

```mlir
module  {
  firrtl.circuit "Ops" {
    firrtl.module @Ops(%os: !firrtl.flip<sint<14>>) {
      %c0_si = firrtl.constant(0 : si64) : !firrtl.sint
      firrtl.connect %os, %c0_si : !firrtl.flip<sint<14>>, !firrtl.sint
    }
  }
}
```

```mlir
// firtool -verify-each=0  --lower-to-rtl --print-ir-before-all --enable-lower-types test.fir
rtl.module @Ops() -> (%os: i14) {
  %c0_i64 = rtl.constant(0 : i64) : i64
  %c0_si = firrtl.constant(0 : si64) : !firrtl.sint
  %0 = firrtl.pad %c0_si, 14 : (!firrtl.sint) -> !firrtl.sint<14>
  %1 = firrtl.stdIntCast %0 : (!firrtl.sint<14>) -> i14
  rtl.output %1 : i14
}
```

```
test.fir:2:3: error: 'firrtl.pad' op result type should be '!firrtl.sint'
  module Ops:
  ^
test.fir:2:3: note: see current operation: %0 = "firrtl.pad"(%c0_si) {amount = 14 : i32} : (!firrtl.sint) -> !firrtl.sint<14>
```

seldridge, 2021-02-18:
I think this is in the bucket of "we need a width inference pass". We can get around this for the time being by compiling down to low FIRRTL using the SFC and then using this regression.

lattner, 2021-02-21:
Right.  The -lower-to-rtl pass is producing this error, it assumes that widths have been inferred already (and many other lowerings have taken place).

fabianschuiki, 2021-04-29:
PRs #980 and #981 take a first shot at this for non-aggregate types and a first batch of operations.

fabianschuiki, 2021-06-21:
Closed as per #1014.

closed 2021-06-21 by fabianschuiki