llvm/circt #615: [FIRRTL] (LowerFIRRTLTypes) cpu / memory anomaly

drom, 2021-02-18:
Running synthetic benchmark `test1` : https://github.com/circt/perf/blob/trunk/regress/test1.fir

Looking at the  execution timing report : https://github.com/circt/perf/blob/gh-pages/test1-2021-02-18.log

```
===-------------------------------------------------------------------------===
                      ... Pass execution timing report ...
===-------------------------------------------------------------------------===
  Total Execution Time: 22.6831 seconds

   ---User Time---   ---Wall Time---  --- Name ---
   0.1536 (  0.7%)     0.1536 (  0.7%)  CSE
   0.0035 (  0.0%)     0.0035 (  0.0%)    (A) DominanceInfo
   1.1773 (  5.1%)     1.1773 (  5.2%)  Canonicalizer
  18.3362 ( 78.9%)    18.3362 ( 80.8%)  'firrtl.circuit' Pipeline
  18.3362 ( 78.9%)    18.3362 ( 80.8%)    'firrtl.module' Pipeline
  18.3362 ( 78.9%)    18.3362 ( 80.8%)      LowerFIRRTLTypes
   0.1732 (  0.7%)     0.1732 (  0.8%)  LowerFIRRTLToRTLModule
   1.2096 (  5.2%)     0.6471 (  2.9%)  'rtl.module' Pipeline
   0.8996 (  3.9%)     0.4691 (  2.1%)    LowerFIRRTLToRTL
   0.3100 (  1.3%)     0.1780 (  0.8%)    AlwaysFusion
   0.5507 (  2.4%)     0.5507 (  2.4%)  CSE
   0.0823 (  0.4%)     0.0823 (  0.4%)    (A) DominanceInfo
   1.6451 (  7.1%)     1.6451 (  7.3%)  Canonicalizer
  23.2457 (100.0%)    22.6831 (100.0%)  Total

{ totalTime: 24.228, maxMemory: 5129453568 }
```

I observe that LowerFIRRTLTypes pass takes 80% of compile time, and probably allocates most of the 5GB memory.

The Memory usage graph over time looks like this :
[image]

nightly runs : https://circt.github.io/perf/

lattner, 2021-02-20:
@mikeurbach  FYI,  I assume this is something algorithmic in LowerTypes that is easy to fix, but I haven't taken a look

mikeurbach, 2021-02-22:
Thanks for the report, this is awesome. I will take a look at the performance in LowerTypes.

mikeurbach, 2021-02-23:
I've been taking a look at this a bit, and I think there are some algorithmic improvements that are pretty low hanging fruit. For starters, I've been looking at CPU flame graphs. Here's a representative one from the latest of the main branch:

[image]

If you click into that, you can click around and see where the CPU time is spent. Interestingly, what jumps out to me are the calls to `mlir::NamedAttrList::getDictionary`, within calls to various helpers to get and set attributes on ops.

I think a lot of these can be avoided, and I'll start sharing patches to avoid doing unnecessary work with the modules' attributes. After those are cleaned up, I will take a look at the memory issue, if it persists.

drom, 2021-02-23:
Nice flame graph. Is it on the `test1`?

seldridge, 2021-02-23:
@mikeurbach: How are you generating that flamegraph? (I'd like to do the same.)

mikeurbach, 2021-02-23:
Yep. I traced the execution using `perf`.

mikeurbach, 2021-02-23:
I ran the test like this:

```
perf record -F 120 -g build/bin/firtool ../perf/regress/test1.fir -format=fir -lower-to-rtl -enable-lower-types -pass-timing > /dev/null
```

That creates a file `perf.data`. Then I used the tools and instructions [here](http://www.brendangregg.com/perf.html#FlameGraphs) to process that into a flame graph:

```
perf script | ../FlameGraph/stackcollapse-perf.pl | ../FlameGraph/flamegraph.pl > perf-original.svg
```

seldridge, 2021-02-23:
Thanks @mikeurbach!

drom, 2021-02-23:
@mikeurbach I do see improvements in the nightly build
Memory 5G -> 3.5G
Exec time 16s -> 12s
https://circt.github.io/perf/

lattner, 2021-03-03:
Hey @drom, what does the time/mem look like with Mike's recent PR 651?  It was:
Memory 5G -> 3.5G
Exec time 16s -> 12s

lattner, 2021-03-03:
Memory 5G -> 3.5G
Exec time 16s -> 12s
https://circt.github.io/perf/

lattner, 2021-03-04:
I just landed 0e7f4cd, which changes this:

```
 Total Execution Time: 85.7807 seconds

   ---User Time---   ---Wall Time---  --- Name ---
   1.4308 (  1.1%)     1.4308 (  1.7%)  CSE
   0.0318 (  0.0%)     0.0318 (  0.0%)    (A) DominanceInfo
  12.3671 (  9.2%)    12.3671 ( 14.4%)  Canonicalizer
   2.4204 (  1.8%)     2.4204 (  2.8%)  LowerFIRRTLToRTLModule
  105.7211 ( 79.0%)    57.6713 ( 67.2%)  'rtl.module' Pipeline
   5.9214 (  4.4%)     2.6246 (  3.1%)    LowerFIRRTLToRTL
  99.7997 ( 74.6%)    55.0467 ( 64.2%)    RTLCleanup
   1.9074 (  1.4%)     1.9074 (  2.2%)  CSE
   0.1140 (  0.1%)     0.1140 (  0.1%)    (A) DominanceInfo
   9.9836 (  7.5%)     9.9836 ( 11.6%)  Canonicalizer
  133.8305 (100.0%)    85.7807 (100.0%)  Total
```

into this:

```
  Total Execution Time: 28.1047 seconds

   ---User Time---   ---Wall Time---  --- Name ---
   1.4178 (  4.0%)     1.4178 (  5.0%)  CSE
   0.0292 (  0.1%)     0.0292 (  0.1%)    (A) DominanceInfo
  10.3369 ( 29.1%)    10.3369 ( 36.8%)  Canonicalizer
   1.3025 (  3.7%)     1.3025 (  4.6%)  LowerFIRRTLToRTLModule
  13.1260 ( 36.9%)     5.6652 ( 20.2%)  'rtl.module' Pipeline
   8.7525 ( 24.6%)     3.7467 ( 13.3%)    LowerFIRRTLToRTL
   4.3735 ( 12.3%)     1.9185 (  6.8%)    RTLCleanup
   1.8085 (  5.1%)     1.8085 (  6.4%)  CSE
   0.1050 (  0.3%)     0.1050 (  0.4%)    (A) DominanceInfo
   7.5737 ( 21.3%)     7.5737 ( 26.9%)  Canonicalizer
  35.5655 (100.0%)    28.1047 (100.0%)  Total
```

This is a debug build though, so all timings are dubious.

drom, 2021-03-04:
Great news! And the memory went 3.3G -> 140M
Tomorrow I will need to add the test2 !

lattner, 2021-03-04:
Oh noe!  It will never stop ;-)

drom, 2021-03-05:
Great! I like it

closed 2021-03-05 by drom

lattner, 2021-03-15:
Just FYI, the next merge of LLVM should [significantly reduce canonicalizer time](https://llvm.discourse.group/t/speeding-up-canonicalize/3015/13).  River is working on further improvements as well.