llvm/circt #617: [RTL] Instance should verify that its return types match the Module

youngar, 2021-02-19:
RTL does not appear to be checking that the number of arguments returned by InstanceOp match the module it is referencing.  The following does not produce an error:

```mlir
// RUN: circt-opt ./test.mlir

rtl.module @empty() -> () {
  rtl.output
}

rtl.module @test() -> () {
  // too many outputs
  %0, %1, %3 = rtl.instance "inline-me" @mem() : () -> (i2, i2, i2)
  rtl.output
}
```

85ayush, 2021-02-20:
I would love to work on fixing this bug. I am a beginner to open source contribution(basic knowledge of git and git hub), can you please guide me and I will give my 100%. I am good at c++. Currently I am studying in NIT raipur, and have got selected as a software engineer at CGI inc.

youngar, 2021-02-20:
Hi @85ayush, thank you for your interest but this is already being worked on over [here](https://github.com/llvm/circt/pull/622).

PR #622 (2021-02-22): [RTL] Add result type verification of instance op
This PR adds the following verification for instance op result types. 

* The numbers of results are the same
* Result types are the same (except for none type)

Will close #617
closed 2021-02-22 by uenoku