llvm/circt #621: [SV] ifdef fusion

prithayan, 2021-02-19:
ifdef with same conditions need to be fused.  This is similar to alwaysFusion logic, https://github.com/llvm/circt/pull/544, but the cases we need to handle are different.
We need to discuss which cases are legal. Here are some of the conditions we need to consider, 

Case 1: Two consecutive ifdefs with same condition should be fused.
```
rtl.module  @test1(%arg0: i1, %arg1: i1) {
  sv.ifdef "!SYNTHESIS"  {
    sv.fwrite "Yes"
  }
  sv.ifdef "!SYNTHESIS"  {
    sv.fwrite "RTE"
  }
  rtl.output
}
```
Case 2: 
ifdef with same condition within an always block should be fused
Note that the always block is an SSACFG region. (Current alwaysFusion logic cannot handle this.)
```
rtl.module  @test1(%arg0: i1, %arg1: i1) {
  sv.always posedge  %arg0 {
    sv.ifdef "!SYNTHESIS"  {
      sv.fwrite "Yes"
    }
    sv.ifdef "!SYNTHESIS"  {
      sv.fwrite "RTE"
    }
  }
  rtl.output
}
```
Case 3:
Nested ifdefs with same condition can be removed.
```
rtl.module  @test1(%arg0: i1, %arg1: i1) {
  sv.ifdef "!SYNTHESIS"  {
    sv.fwrite "Yes"
    sv.ifdef "!SYNTHESIS"  {
      sv.fwrite "RTE"
    }
  }
  rtl.output
}
```
Case 4: 
ifdef within a different condition cannot be fused. 

```
rtl.module  @test1(%arg0: i1, %arg1: i1) {
  sv.ifdef "!SYNTHESIS"  {
    sv.fwrite "Yes"
  }
  sv.ifdef "RANDOM"  {
    sv.ifdef "!SYNTHESIS"  {
      sv.fwrite "NO"
    }
  }
  rtl.output
}
```

Case 5:
Non-consecutive ifdefs with same condition can be fused. 
```
rtl.module  @test1(%arg0: i1, %arg1: i1) {
  sv.ifdef "!SYNTHESIS"  {
    sv.fwrite "Yes"
  }
  sv.ifdef "RANDOM"  {
    sv.fwrite "RANDOM"
  }
  sv.ifdef "!SYNTHESIS"  {
    sv.fwrite "NO"
  }
  rtl.output
}     
```
Case 6:
Same as above, Since these are SSA values, it is legal to fuse the following ifdefs  with same condition. After fusion, the values need to be renamed.
```
rtl.module  @test1(%arg0: i1, %arg1: i1) {
  %0 = sv.textual_value "PRINTF_COND_" : i1
  sv.ifdef "!SYNTHESIS"  {
    %1 = comb.and %arg0, %arg1 : i1
  }
  sv.ifdef "RANDOM"  {
    %1 = comb.add %0, %arg1 : i1
  }
  sv.ifdef "!SYNTHESIS"  {
    %1 = comb.xor %0, %arg1 : i1
  }
  rtl.output
}
```

seldridge, 2021-02-19:
Case 4 could be fused if we allow for re-ordering the `sv.ifdef` (possibly out of scope of the fusion discussed here.

Case 5 is tricky because fusing that will reorder output statements when `RANDOM` is defined. A similar thing would be if the middle `ifdef` is doing a blocking assign where re-ordering the statements would affect result. I'm not sure how to properly generalize what I'm describing here.

youngar, 2021-02-19:
Case 5 looks tricky, but since it is a `rtl.module` is a graph region it should theoretically be "legal" to merge?  Maybe I'm leaning too heavily on the graph region concept, but possibly it would be the IR that is wrong, not the transformation? Side note, maybe it would make sense to print Graph Regions differently?

For case 6, value naming `%1` is handled by the printer, it's not something you have to worry about when doing IR transformations.

And I might add:

if def squashing - condition is always true, possibly a canonicalization?
```mlir
rtl.module  @test1(%arg0: i1, %arg1: i1) {
  sv.ifdef "RANDOM"  {
    sv.ifdef "RANDOM"  {
    	%1 = comb.add %0, %arg1 : i1
  	}
    %1 = comb.add %0, %arg1 : i1
  }
}
```

SSA Region - no side effecting operations:
```mlir
ssacfg-region {
  sv.ifdef "!SYNTHESIS"  {
    %0 = comb.and %arg0, %arg1 : i1
  }
  %0 = comb.and %arg0, %arg1 : i1
  sv.ifdef "!SYNTHESIS"  {
    %1 = comb.xor %arg0, %arg1 : i1
  }
}
```

SSA Region - with side effects blocking merging:
```mlir
ssacfg-region {
  sv.ifdef "!SYNTHESIS"  {
    %0 = comb.and %arg0, %arg1 : i1
  }
  sv.fwrite "SYTHESIS = 1" // Side effecting operation
  sv.ifdef "!SYNTHESIS"  {
    %1 = comb.xor %arg0, %arg1 : i1
  }
}
```

lattner, 2021-02-21:
I implemented some support for this in the generalized alwaysfusion pass (now known as rtl cleanup), and eliminated the "!FOO" form of conditionals which defines away some of the complexity.

lattner, 2021-02-21:
I think this is done now, other than the scoping based transformation of "ifdef X with ifdef X inside of it can be simplified".  I don't think this will occur with the Chisel frontend though, so I think we're ok without it for the forseeable future.

prithayan, 2021-02-22:
Closing this issue, Fixed as mentioned above in https://github.com/llvm/circt/commit/ee01f0817b3712cc849d4f539c37b26d12f6004b

closed 2021-02-22 by prithayan