llvm/circt #623: [SV] Expose memory control signals for interfacing

JuanEsco063, 2021-02-19:
While working with @mikeurbach to finish the details of implementing memories as discussed in #543 I am faced with the issue of interfacing the RTL code generated with a more general design, specially for loading data on the memories and reading back from them.

One way I have been able to achieve this (at least in the Xilinx architecture) is taking advantage of the dual port BRAM primitives. I have manually tweaked the RTL code generated and exposed the RAM control signals from inside the modules all the way up as input/outputs of the top level module. Then I control one of the ports of the BRAM modules with an external processor using an AXI BRAM controller and the other port is controlled by the RTL circuit. This allows me to load the data I need on the memories, and read back once the results have been calculated by the accelerator.

My question is: is there an easy way to expose the memory control signals (load address[output], write address[output], load data[input], write data[output], and write enable [output]) as input/output of the top level module? Alternatively, is there a roadmap for integration with external circuitry for modules involving memories?

lattner, 2021-02-20:
Hey Juan,

I'm not exactly sure here, but I think this could a good topic for discussion at a wednesday design meeting.  

@stephenneuendorffer  just FYI

JuanEsco063, 2021-02-22:
@lattner Great idea! Just added a comment on the shared document. Since I see a few points on the agenda I am not going to close this issue just yet in case we don't get to talk about it this Wednesday.