llvm/circt #627: [SV] Split sv.ifdef into two different operations

lattner, 2021-02-20:
It came up in a discussion yesterday that sv.ifdef should be split into the different nodes, one that is a `ProceduralRegion` that can be used in an always block, and one that is a graph region that can be used in a top level of an rtl.module.

This will allow us to build better assertions, e.g. check that a fwrite only occurs in `ProceduralRegion`'s

teqdruid, 2021-02-20:
Why not do a recursive check that ops are in a procedural region?

lattner, 2021-02-20:
Splitting it this way just makes it simpler to implement the checks, and makes them local.

teqdruid, 2021-02-20:
Fair enough. I just worry about havIng to split a bunch of ops in this
manner, but this is probably the practical choice if it’s just one or a
small number.

lattner, 2021-02-20:
Yep, I think it is just ifdef

lattner, 2021-02-21:
I'm going to take this, as this will simplify other things in my path

lattner, 2021-02-21:
Done in 906690b

closed 2021-02-21 by lattner