llvm/circt #629: [ESI] [Cosim] Question about Endpoint API design.

sequencer, 2021-02-20:
Hi, @teqdruid  I'm trying to experimentally integrate Chisel(Scala) API for Cosim.

I have a question about the decouple logic in endpoint design.
Cosim use the ready-valid decoupled signal for software<->simulator message transmission. My question is how to design this ready-valid signal pair?

From my understanding to the design of simulator, a DPI call is guarded by it's `clock`, `reset`, `condition`. This `condition` is a signal from internal circuit state, while in Cosim, it seems this `condition` are `&` from both circuit state with `valid` signal and software state `ready` signal. I think during simulator(i.e. verilator) running, simulator cannot retrieve `ready` signal from software side, while software cannot retrieve `valid` signal from simulator, since overhead is too heavy.
The only usage which I can think about is using Cosim with FPGA, inserting hardware queue between software and DUT on FPGA.

Just curious about this design purpose :)

teqdruid, 2021-02-21:
> Cosim use the ready-valid decoupled signal for software<->simulator message transmission. My question is how to design this ready-valid signal pair?

The easiest way is to take the `!esi.channel` which goes to/from `esi.cosim` and use `esi.unwrap.vr` and/or `esi.wrap.vr` to get the data, valid, and ready signals. You can then adapt those signals pretty easily to a latency insensitive (aka LI or elastic) module. If you're trying to connect it to a non-LI module interface, that'll take a bit more thought.

I don't know much about Chisel, but here's what I suggest in general. If you're thinking about how to expose a cosim interface up to a language, I'd suggest writing a cosim API which exposes the cosim channels using the appropriate language facilities: in HLS, it's be some sort of streams. In a language without built in support for something like streams, a ready-valid-data interface may be your best bet. It'd be great to have support for ESI cosim in Chisel! I'd love to discuss this potential more, but this probably isn't the best venue for a longer discussion. I'd prefer the [discourse forum](https://llvm.discourse.group/c/Projects-that-want-to-become-official-LLVM-Projects/circt/).

> From my understanding to the design of simulator, a DPI call is guarded

Are you referring to DPI in simulators in general or the ESI cosim DPI calls? If in general, a DPI call is not guarded at all. It can be called at any point just like an ordinary SystemVerilog function call. The ESI cosim DPI calls (e.g. [here](https://github.com/llvm/circt/blob/5ad5992b673587e421a39abf9c28f8e2f40c0d39/include/circt/Dialect/ESI/cosim/Cosim_Endpoint.sv#L78)) are made at each clock and mostly when reset is not being asserted.

From the DPI plugin's point of view, it's just a normal function call with accessors to get at array data. If the simulator is multi-threaded, it is the simulators responsibility to ensure thread safety of the data accessible to each DPI function call through its arguments.

> I think during simulator(i.e. verilator) running, simulator cannot retrieve `ready` signal from software side, while software cannot retrieve `valid` signal from simulator, since overhead is too heavy.

I think you misunderstand what ESI cosim is doing. ESI cosim only sends ESI _messages_, not the wire-level signals. At the wire level, message is defined by both valid and ready being asserted on the same clock cycle. The valid and ready signals are not sent, just the data as a logical message.

You are correct that transmitting raw bits every clock cycle would be too heavy weight for the mechanisms which ESI cosim employs.

> The only usage which I can think about is using Cosim with FPGA, inserting hardware queue between software and DUT on FPGA

Cosimulation can theoretically be used anywhere a simulation is being run, not just FPGAs. In the future, the thought is that we'd expose the same ESI ports (channels) API to simulation and synthesized hardware. None of this is FPGA specific, though I'm mostly targeting FPGAs initially. The ESI port abstraction and the corresponding software API would be used to abstract away the board support package interface (the communication mechanism) from the communication channels to make designs more portable. One could even imagine using this on ASICs with the same purpose -- abstracting away the details / mechanics of communication.

I have yet to adapt and merge the ESI "rationale" document into CIRCT, though I'll get around to it soon since I'm in writing mode currently.

sequencer, 2021-02-21:
@teqdruid Thanks for your explanation! Sorry, I misunderstood the design purpose of ESI, I used to think it's only designed for simulation only. After reading [this](https://llvm.discourse.group/t/elastic-silicon-interconnect-esi-modeling/1992), I understand it now. 

My design purpose is borrowing infrastructure of cosim, adopt which to bridge arbitrary interface between Chisel bundle and software, I really like the way you use capnp! I'm going to use the hardware signal to guard DPI, then simulator no need to schedule DPI call each cycle to speed up simulation.

I'll learn from Cosim code and figure out a way to adopt as the simulation infra in Chisel. Thank you!

closed 2021-02-21 by sequencer