llvm/circt #630: [RTL] rename RTL dialect to HW?

lattner, 2021-02-20:
John [moved](https://github.com/llvm/circt/pull/599) the mechanics of the RTL dialect out to "comb".  The things we have left are rtl.module, types, and a few things for working with aggregates.  The name "RTL" isn't really applicable any more.

What is left is now the macro structure of the CIRCT project that I think will be the unifying theory of a bunch of interesting work.  I think we can get away with renaming this to "hw" (as in "hardware").  Any other suggestions?

I also considered "circt" as the dialect name, but that will be confusing as the CIRCT project grows to include more stuff.

teqdruid, 2021-02-21:
> I also considered "circt" as the dialect name, but that will be confusing as the CIRCT project grows to include more stuff.

I agree.

> What is left is now the macro structure of the CIRCT project that I think will be the unifying theory of a bunch of interesting work. I think we can get away with renaming this to "hw" (as in "hardware"). Any other suggestions?

I like `hw`. I also like `core` as in `circt.core`.

lattner, 2021-02-21:
Which do you think is better?  I'll add it to the wednesday agenda.

teqdruid, 2021-02-21:
I prefer `circt.core`, but then my first compiled language was Java which tends to use `core` in package names a lot.

lattner, 2021-02-24:
Per discussion today in the design meeting, we agreed to go with "hw" for the dialect, and look into whether it makes sense to use "circt::hw::ModuleOp" and "circt::hw::ArrayType" for the C++ types or to flatten into "circt::HWModuleOp"

lattner, 2021-05-15:
This is effectively done.

closed 2021-05-15 by lattner