llvm/circt #640: FIRRTLToRTL test fails with gcc 10.2 but not with clang 11

teqdruid, 2021-02-22:
```
FAIL: CIRCT :: Conversion/FIRRTLToRTL/lower-to-rtl.mlir (42 of 180)
******************** TEST 'CIRCT :: Conversion/FIRRTLToRTL/lower-to-rtl.mlir' FAILED ********************
Script:
--
: 'RUN: at line 1';   /home/jodemme/circt/build/bin/circt-opt -lower-firrtl-to-rtl /home/jodemme/circt/test/Conversion/FIRRTLToRTL/lower-to-rtl.mlir | /home/jodemme/circt/llvm/build/bin/FileCheck /home/jodemme/circt/test/Conversion/FIRRTLToRTL/lower-to-rtl.mlir
--
Exit Code: 1

Command Output (stderr):
--
/home/jodemme/circt/test/Conversion/FIRRTLToRTL/lower-to-rtl.mlir:801:12: error: CHECK: expected string not found in input
 // CHECK: %[[mask:.+]] = sv.read_inout %[[mask_0]]
           ^
<stdin>:423:25: note: scanning from here
 %15 = sv.read_inout %9 : !rtl.inout<i1>
                        ^
<stdin>:423:25: note: with "mask_0" equal to "11"
 %15 = sv.read_inout %9 : !rtl.inout<i1>
                        ^
<stdin>:426:2: note: possible intended match here
 %17 = sv.read_inout %10 : !rtl.inout<i4>
 ^

Input file: <stdin>
Check file: /home/jodemme/circt/test/Conversion/FIRRTLToRTL/lower-to-rtl.mlir

-dump-input=help explains the following input dump.

Input was:
<<<<<<
             .
             .
             .
           418:  sv.passign %12, %17 : i8
           419:  }
           420:  }
           421:  sv.alwaysff(posedge %8) {
           422:  %14 = sv.read_inout %11 : !rtl.inout<i1>
           423:  %15 = sv.read_inout %9 : !rtl.inout<i1>
check:801'0                             X~~~~~~~~~~~~~~~ error: no match found
check:801'1                                              with "mask_0" equal to "11"
           424:  %16 = comb.and %15, %14 : i1
check:801'0     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
           425:  sv.if %16 {
check:801'0     ~~~~~~~~~~~~
           426:  %17 = sv.read_inout %10 : !rtl.inout<i4>
check:801'0     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
check:801'2      ?                                        possible intended match
           427:  %18 = sv.array_index_inout %memory[%17] : !rtl.inout<uarray<16xi8>>, i4
check:801'0     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
           428:  %19 = sv.read_inout %12 : !rtl.inout<i8>
check:801'0     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
           429:  sv.passign %18, %19 : i8
check:801'0     ~~~~~~~~~~~~~~~~~~~~~~~~~
           430:  }
check:801'0     ~~
           431:  }
check:801'0     ~~
             .
             .
             .
>>>>>>

--

********************
********************
Failed Tests (1):
  CIRCT :: Conversion/FIRRTLToRTL/lower-to-rtl.mlir

Testing Time: 1.54s
  Passed: 179
  Failed:   1
```

Seems like the two lines are reverse with gcc. I suspect the recent change to ExportVerilog may have some compiler specific behavior...?

teqdruid, 2021-02-22:
This appears in @seldridge's code, but I don't think his commit is to blame.

seldridge, 2021-02-22:
Ugh... Quickly checking... I think function argument evaluation order is undefined. GCC is likely doing something with a stack here while LLVM isn't. 

The simplest fix (which I'll push) is to switch this to a `CHECK-DAG` as the order of those truly doesn't matter.

lattner, 2021-02-22:
Yeah, another likely possibility is if something is relying on non-determinstic things.  A classic example is depending on iteration order in a DenseMap since the elements are in a hash order.

seldridge, 2021-02-22:
I thought it might be iteration order, but I'm pretty sure this is just argument order: https://github.com/llvm/circt/blob/c9ef5a24dd94f692ecab5aedcf1d1b16f5ddd1ac/lib/Conversion/FIRRTLToRTL/LowerToRTL.cpp#L1585

```c++
        auto cond = builder->create<comb::AndOp>(
            builder->create<sv::ReadInOutOp>(last.en),
            builder->create<sv::ReadInOutOp>(last.mask));
```

lattner, 2021-02-23:
Yes, you're totally right.  i'll fix Thx!

lattner, 2021-02-23:
Fixed in 921e8e7, does that resolve the issue overall?

seldridge, 2021-02-23:
@lattner: I may have scooped you on this with c97f6689 and 5be0ad26. @darthscsi found a different, similar issue that 5be0ad26 was supposed to address due to the same thing going on here: https://github.com/llvm/circt/blob/921e8e7501b50990d3dabdcffee2a262581ccae6/lib/Conversion/FIRRTLToRTL/LowerToRTL.cpp#L1273.

youngar, 2021-02-23:
Tangentially related, I opened an issue about adding GCC builds here: https://github.com/llvm/circt/issues/644

lattner, 2021-02-23:
@seldridge please revert the CHECK-DAG changes.  It is important that the compiler be deterministic here.  We should drop in code changes to make it so, instead of relaxing the tests.  thx!

lattner, 2021-02-23:
I agree the BPAssignOp line on 1273 also needs to be fixed.  C++ is a terrible terrible language :-(

seldridge, 2021-02-23:
I'll revert.

I had originally thought that this doesn't matter (either is functionally correct). However, if those are temporaries that are propagated to the Verilog, then you may wind up with `_T_0` for GCC and `_T_1` for LLVM. This is a real mess and needs to be avoided.

lattner, 2021-02-23:
Thanks!  It is really important for compilers to be determinstic.  Not doing so causes all sorts of issues, incl when you're trying to binary search bugs etc.

seldridge, 2021-02-23:
The line 1273 issue is fixed in 1329169a.

darthscsi, 2021-03-26:
I believe this is resolved.  I can't reproduce on ToT and I think the arg order instances are all handled now.

closed 2021-03-26 by darthscsi