llvm/circt #661: [LowerTypes]

darthscsi, 2021-02-24:
memories with multiple ports are not lowered correctly.  This segfaults:
```
module  {
  firrtl.circuit "foo" {
    firrtl.module @foo(
      %clock: !firrtl.clock
      ) {
      %head_MPORT_2, %head_MPORT_6 = firrtl.mem Undefined {depth = 20 : i64, name = "head", portNames = ["MPORT_2", "MPORT_6"], readLatency = 0 : i32, writeLatency = 1 : i32} 
      : !firrtl.flip< bundle<addr: uint<5>, en: uint<1>, clk: clock, data: uint<5>, mask: uint<1> > >,
        !firrtl.flip< bundle<addr: uint<5>, en: uint<1>, clk: clock, data: uint<5>, mask: uint<1> > >
      %127 = firrtl.subfield %head_MPORT_6("clk") : (!firrtl.flip<bundle<addr: uint<5>, en: uint<1>, clk: clock, data: uint<5>, mask: uint<1>>>) -> !firrtl.flip<clock>
    }
  }
}
```

seldridge, 2021-02-24:
Digging into this now... The one weird thing I see is that this is a read from a memory input port (which will only be cleaned up once this is lowered to RTL).

darthscsi, 2021-02-24:
A minor note: if a memory has only one field, the pass is still replacing it with an identical memory.

PR #662 (2021-02-24): Fix Inappropriate Use of StringRef in LowerTypes
Fixes #661 (I think?).

Inside FIRRTL's LowerTypes transform, when handling MemOps, I was using a `DenseMap<StringRef, Value>` to track new wire creation where new wires are keyed on the name of the wire. However, this is an inappropriate use of a `StringRef` as the names are going in and out of scope. E.g., I'm doing something like:

```c++
DenseMap<StringRef, Value> newWires;
newWires["foo"];
newWires["foo"];
```

I should be using something where the key is permanent and not some transient memory address.

This PR changes this from `DenseMap< [... cut]
closed 2021-02-24 by seldridge