llvm/circt #667: Always blocks not being merged

darthscsi, 2021-02-24:
I think location attribute is preventing merging of always blocks by the cleanup pass.  I haven't looked into it.
```
      %1149 = firrtl.and %1138, %543 : (!firrtl.uint<1>, !firrtl.uint<1>) -> !firrtl.uint<1>
      %1150 = firrtl.and %1149, %830 : (!firrtl.uint<1>, !firrtl.uint<1>) -> !firrtl.uint<1>
      firrtl.printf %clock, %1150, "Ch:A Op:Hint           "
      %1151 = firrtl.and %1138, %204 : (!firrtl.uint<1>, !firrtl.uint<1>) -> !firrtl.uint<1>
      %1152 = firrtl.and %1151, %830 : (!firrtl.uint<1>, !firrtl.uint<1>) -> !firrtl.uint<1>
      firrtl.printf %clock, %1152, "Ch:A Op:AcquireBlock   "
      %1153 = firrtl.and %1138, %233 : (!firrtl.uint<1>, !firrtl.uint<1>) -> !firrtl.uint<1>
      %1154 = firrtl.and %1153, %830 : (!firrtl.uint<1>, !firrtl.uint<1>) -> !firrtl.uint<1>
      firrtl.printf %clock, %1154, "Ch:A Op:AcquirePerm    "
```
Output from firtool:
```
  always @(posedge clock) begin	// u77.lo.fir.mlir:7490:7
    `ifndef SYNTHESIS	// u77.lo.fir.mlir:7490:7
      if (`PRINTF_COND_ & _T_687 & _T_225)	// u77.lo.fir.mlir:7490:7
        $fwrite(32'h80000002, "Ch:C Op:AccessAckData  ");	// u77.lo.fir.mlir:7490:7
    `endif
  end // always @(posedge)
  wire _T_688 = _T_685 & _T_262;	// u77.lo.fir.mlir:7491:15
  always @(posedge clock) begin	// u77.lo.fir.mlir:7493:7
    `ifndef SYNTHESIS	// u77.lo.fir.mlir:7493:7
      if (`PRINTF_COND_ & _T_688 & _T_225)	// u77.lo.fir.mlir:7493:7
        $fwrite(32'h80000002, "Ch:C Op:HintAck        ");	// u77.lo.fir.mlir:7493:7
    `endif
  end // always @(posedge)
  wire _T_689 = _T_685 & io_in_c_bits_opcode == 3'h3;	// u77.lo.fir.mlir:5500:15, :5930:15, :7494:15
  always @(posedge clock) begin	// u77.lo.fir.mlir:7496:7
    `ifndef SYNTHESIS	// u77.lo.fir.mlir:7496:7
      if (`PRINTF_COND_ & _T_689 & _T_225)	// u77.lo.fir.mlir:7496:7
        $fwrite(32'h80000002, "Ch:C Op:Invalid Opcode ");	// u77.lo.fir.mlir:7496:7
    `endif
  end // always @(posedge)

```

lattner, 2021-02-25:
interesting, the location are included as part of the hash, see the OperationEquivalence:: methods in OperationSupport.cpp in mlir.  Are these using always_ff nodes?

darthscsi, 2021-02-25:
No, these are always blocks produced from lowering printf.

closed 2021-03-01 by prithayan