llvm/circt #668: ExportVerilog not splitting line for long variadic ops

darthscsi, 2021-02-24:
Not splitting single expressions which are long.
```
    %3844 = comb.or %3233, %3234, %3235, %3236, %3237, %3238, %3240, %3241, %3242, %3243, %3244, %3245, %3246, %3247, %3248, %3249, %3250, %3251, %3252, %3253, %3254, %3255, %3256, %3257, %3258, %3259, %3260, %3261, %3263, %3264, %3265, %3267, %3268, %3269, %3270, %3271, %3272, %3273, %3274, %3275, %3276, %3277, %3278, %3279, %3280, %3281, %3282, %3283, %3284, %3285, %3286, %3288, %3290, %3291, %3292, %3293, %3294, %3295, %3296, %3297, %3298, %3299, %3300, %3301, %3302, %3303, %3304, %3305, %3306, %3307, %3308, %3309, %3310, %3311, %3312, %3313, %3314, %3315, %3316, %3317, %3319, %3320, %3321, %3322, %3323, %3324, %3325, %3326, %3327, %3328, %3329, %3330, %3331, %3332, %3333, %3334, %3335, %3336, %3338, %3339, %3340, %3341, %3342, %3343, %3344, %3345, %3346, %3347, %3349, %3350, %3351, %3352, %3353, %3354, %3355, %3356, %3357, %3358, %3359, %3360, %3362, %3363, %3364, %3365, %3366, %3367, %3368, %3369, %3370, %3371, %3372, %3373, %3374, %3375, %3376, %3377, %3378, %3379, %3380, %3381, %3382, %3383, %3384, %3385, %3386, %3387, %3388, %3390, %3391, %3392, %3393, %3394, %3395, %3396, %3397, %3398, %3399, %3400, %3401, %3402, %3403, %3404, %3405, %3406, %3407, %3408, %3409, %3411, %3412, %3413, %3414, %3415, %3416, %3417, %3418, %3419, %3420, %3421, %3422, %3423, %3424, %3425, %3426, %3427, %3428, %3429, %3430, %3431, %3432, %3433, %3434, %3435, %3436, %3437, %3438, %3439, %3440, %3441, %3442, %3444, %3446, %3447, %3448, %3449, %3450, %3451, %3452, %3453, %3455, %3456, %3457, %3458, %3459, %3460, %3461, %3462, %3463, %3464, %3465, %3466, %3467, %3468, %3469, %3470, %3471, %3472, %3473, %3474, %3475, %3476, %3477, %3478, %3479, %3480, %3481, %3482, %3483, %3484, %3485, %3486, %3487, %3488, %3489, %3490, %3491, %3492, %3493, %3494, %3495, %3496, %3497, %3498, %3499, %3500, %3501, %3502, %3503, %3504, %3505, %3506, %3507, %3508, %3509, %3510, %3511, %3512, %3513, %3514, %3515, %3516, %3517, %3518, %3519, %3520, %3521, %3522, %3523, %3524, %3525, %3526, %3527, %3528, %3529, %3530, %3531, %3532, %3533, %3534, %3535, %3536, %3537, %3538, %3539, %3540, %3541, %3542, %3543, %3544, %3545, %3546, %3547, %3548, %3549, %3550, %3551, %3552, %3553, %3554, %3555, %3556, %3557, %3558, %3559, %3560, %3561, %3562, %3563, %3564, %3565, %3566, %3567, %3568, %3569, %3570, %3571, %3572, %3573, %3574, %3575, %3576, %3577, %3578, %3579, %3580, %3581, %3583, %3584, %3585, %3586, %3588, %3589, %3590, %3591, %3592, %3593, %3594, %3595, %3596, %3597, %3598, %3599, %3600, %3601, %3602, %3603, %3604, %3605, %3606, %3607, %3608, %3609, %3610, %3611, %3612, %3613, %3614, %3615, %3616, %3617, %3618, %3619, %3620, %3621, %3622, %3623, %3624, %3625, %3626, %3627, %3628, %3629, %3630, %3631, %3632, %3633, %3634, %3635, %3636, %3637, %3638, %3639, %3640, %3641, %3642, %3643, %3644, %3645, %3646, %3647, %3648, %3649, %3650, %3651, %3652, %3653, %3654, %3655, %3656, %3657, %3658, %3659, %3660, %3661, %3662, %3663, %3664, %3665, %3666, %3667, %3668, %3669, %3670, %3671, %3672, %3673, %3674, %3675, %3676, %3677, %3678, %3679, %3680, %3681, %3682, %3683, %3684, %3685, %3686, %3687, %3688, %3689, %3690, %3691, %3692, %3693, %3694, %3695, %3696, %3697, %3698, %3699, %3700, %3701, %3702, %3703, %3704, %3705, %3706, %3707, %3708, %3709, %3710, %3711, %3712, %3713, %3714, %3715, %3716, %3717, %3718, %3719, %3720, %3721, %3722, %3723, %3724, %3725, %3726, %3727, %3728, %3729, %3730, %3731, %3732, %3733, %3734, %3735, %3736, %3737, %3738, %3739, %3740, %3741, %3742, %3743, %3744, %3745, %3746, %3747, %3748, %3749, %3750, %3751, %3752, %3753, %3754, %3755, %3756, %3757, %3758, %3759, %3760, %3761, %3762, %3763, %3764, %3765, %3766, %3767, %3768, %3769, %3770, %3771, %3772, %3773, %3774, %3775, %3776, %3777, %3778, %3779, %3780, %3781, %3782, %3783, %3784, %3785, %3786, %3787, %3788, %3789, %3790, %3791, %3792, %3793, %3794, %3795, %3796, %3797, %3798, %3799, %3800, %3801, %3802, %3803, %3804, %3805, %3806, %3807, %3808, %3809, %3810, %3811, %3812, %3813, %3814, %3815, %3816, %3817, %3818, %3819, %3820, %3821, %3822, %3823, %3824, %3825, %3826, %3827, %3828, %3829, %3830, %3831, %3832, %3833, %3834, %3835, %3836, %3837, %3838, %3840, %3841, %3842, %3843 : i1
```
Makes something like:

```
  wire _T_361 = _T_180 ? _T_181 == 64'h63
| _T_181 == 64'h1063
| _T_181 == 64'h4063
| _T_181 == 64'h6063
| _T_181 == 64'h5063
| _T_181 == 64'h7063
| _T_182 == 64'h6F
| _T_181 == 64'h67
| _T_181 == 64'h3
| _T_181 == 64'h4003
| _T_181 == 64'h1003
| _T_181 == 64'h5003
| _T_181 == 64'h2003
| _T_181 == 64'h6003
| _T_181 == 64'h3003
| _T_181 == 64'h23
| _T_181 == 64'h1023
| _T_181 == 64'h2023
| _T_181 == 64'h3023
| _T_182 == 64'h37
| _T_182 == 64'h17
| _T_181 == 64'h13
| _T_181 == 64'h2013
| _T_181 == 64'h3013
| _T_181 == 64'h7013
| _T_181 == 64'h6013
| _T_181 == 64'h4013
| _T_181 == 64'h1B
| _T_183 == 64'h101B
| _T_183 == 64'h501B
| _T_183 == 64'h4000501B
| _T_184 == 64'h1013
| _T_184 == 64'h5013
| _T_184 == 64'h40005013
| _T_183 == 64'h33
| _T_183 == 64'h40000033
| _T_183 == 64'h2033
| _T_183 == 64'h3033
| _T_183 == 64'h7033
| _T_183 == 64'h6033
| _T_183 == 64'h4033
| _T_183 == 64'h1033
| _T_183 == 64'h5033
| _T_183 == 64'h40005033
| _T_183 == 64'h3B
| _T_183 == 64'h4000003B
| _T_183 == 64'h103B
| _T_183 == 64'h503B
| _T_183 == 64'h4000503B
| _T_181 == 64'hF
| _T_181 == 64'h100F
| (io_fetch_bits_data & 64'hFE007FFF) == 64'h12000073
| _T_185 == 64'hFC000073
| _T_185 == 64'hFC200073
| _T_181 == 64'h5073
| _T_181 == 64'h6073
| _T_181 == 64'h7073
| _T_185 == 64'h73
| _T_185 == 64'h100073
| _T_185 == 64'h30200073
| _T_185 == 64'h10500073
| _T_185 == 64'h30500073
| _T_181 == 64'h1073
[... 1693 lines cut ...]
| _T_254 == 32'h80005057
| _T_254 == 32'h84005057
| _T_254 == 32'hB0005057
| _T_254 == 32'hB4005057
| _T_254 == 32'hB8005057
| _T_254 == 32'hBC005057
| _T_254 == 32'hA0005057
| _T_254 == 32'hA4005057
| _T_254 == 32'hA8005057
| _T_254 == 32'hAC005057
| _T_260 == 32'h42005057
| _T_254 == 32'hC0005057
| _T_254 == 32'hC8005057
| _T_254 == 32'hE0005057
| _T_254 == 32'hD0005057
| _T_254 == 32'hD8005057
| _T_254 == 32'hF0005057
| _T_254 == 32'hF4005057
| _T_254 == 32'hF8005057
| _T_254 == 32'hFC005057
| _T_264 == 32'h42002057
| _T_263 == 32'h40082057
| _T_263 == 32'h4008A057
| _T_264 == 32'h42001057;	// u77.lo.fir.mlir:493257:17, :493315:27, :493318:27, :493319:27, :493351:27, :493352:27, :493353:27, :493354:27, :493360:27, :493361:27, :493362:27, :493363:27, :493427:27, :493428:27, :493429:27, :493430:27, :493431:27, :493432:27, :493433:27, :493434:27, :493435:27, :493436:27, :493437:27, :493439:27, :493440:27, :493441:27, :493442:27, :493443:27, :493444:27, :493445:27, :493446:27, :493447:27, :493448:27, :493449:27, :493450:27, :493451:27, :493452:27, :493453:27, :493454:27, :493455:27, :493456:27, :493457:27, :493458:27, :493459:27, :493460:27, :493461:27, :493462:27, :493463:27, :493464:27, :493465:27, :493466:27, :493467:27, :493468:27, :493469:27, :493470:27, :493471:27, :493498:27, :493499:27, :493500:27, :493501:27, :493516:27, :493524:27, :493525:27, :493526:27, :493527:27, :493533:27, :493534:27, :493535:27, :493536:27, :493542:27, :493543:27, :493544:27, :493545:27, :493551:27, :493552:27, :493553:27, :493554:27, :493564:27, :493565:27, :493579:27, :493580:27, :493581:27, :493582:27, :493583:27, :493615:27, :493623:27, :493624:27, :493625:27, :493626:27, :493627:27, :493628:27, :493629:27, :493630:27, :493631:27, :493632:27, :493633:27, :493634:27, :493635:27, :493636:27, :493637:27, :493638:27, :493639:27, :493640:27, :493641:27, :493642:27, :493643:27, :493644:27, :493645:27, :493646:27, :493647:27, :493648:27, :493653:27, :493654:27, :493655:27, :493656:27, :493657:27, :493658:27, :493659:27, :493660:27, :493661:27, :493663:27, :493665:27, :493693:27, :493697:27, :493700:27, :493701:27, :493702:27, :493703:27, :493704:27, :493705:27, :493706:27, :493707:27, :493714:27, :493715:27, :493716:27, :493717:27, :493718:27, :493719:27, :493720:27, :493721:27, :493722:27, :493727:27, :493728:27, :493729:27, :493730:27, :493731:27, :493735:27, :493747:27, :493748:27, :493749:27, :493750:27, :493751:27, :493752:27, :493760:27, :493761:27, :493762:27, :493763:27, :493764:27, :493787:27, :493788:27, :493789:27, :493790:27, :493791:27, :493792:27, :493793:27, :493794:27, :493795:27, :493796:27, :493797:27, :493798:27, :493799:27, :493800:27, :493801:27, :493802:27, :493803:27, :493804:27, :493805:27, :493806:27, :493807:27, :493808:27, :493809:27, :493810:27, :493811:2 [... cut ...] 6:15, :518019:15, :518022:15, :518025:15, :518028:15, :518031:15, :518034:15, :518037:15, :518040:15, :518043:15, :518046:15, :518049:15, :518052:15, :518055:15, :518058:15, :518061:15, :518064:15, :518067:15, :518070:15, :518073:15, :518076:15, :518079:15, :518082:15, :518085:15, :518089:15, :518092:15, :518095:15, :518098:15, :519284:16, :519636:16, :519640:16, :519644:16, :519648:16, :519652:16, :519656:16, :519660:16, :519662:16, :519664:16, :523111:16, :523158:16, :523184:16, :523238:16
```

darthscsi, 2021-02-24:
But all that stuff is on a single line.  This exceeds verilator parser buffer for tokens per line.

lattner, 2021-02-25:
Awesome, let's hear it for line length limitations ;-)

The right thing here is for export-verilog to buffer up subtrees and emit them into temporaries if they exceed some line length.  80 columns ftw?

lattner, 2021-02-25:
Can you attach a self-contained rtl.module example?  I'll take a look at this when I get a chance.

darthscsi, 2021-02-25:
Self contained example: https://github.com/llvm/circt/commit/db921e31ccb7c0b522bcf2e20be85fb92a8d0bdf

lattner, 2021-03-01:
btw, I'm actively working on this.

lattner, 2021-03-01:
Significant progress is in the patches leading up to 51de453

lattner, 2021-03-02:
Done in bf1a3dc

closed 2021-03-02 by lattner