llvm/circt #670: [FIRRTL][RTL][ExportedVerilog] Missing name in black box memory 'rtl.instance' op requires attribute 'instanceName'

JuanEsco063, 2021-02-25:
I am trying to generate RTL for a simple matmul kernel considering memories as black boxes with read and write latency of 1 after #493 and #585 (and modifying the corresponding files to include the changes of #602 ).

After running 

`circt-opt matmul_std_lin.mlir -create-dataflow -handshake-insert-buffer -lower-handshake-to-firrtl > matmul_firrtl_1.mlir `

To generate the FIRRTL dialect file. Then I modify the aforementioned file to include a read latency of 1 and replace memories for their black box equivalent by running:

`circt-opt matmul_firrtl_1.mlir  -firrtl-blackbox-memory > matmul_firrtl_2.mlir`

But when I try to generate Verilog using:

`firtool -format=mlir matmul_firrtl_2.mlir -lower-to-rtl -enable-lower-types -verilog > new_matmul.v`

I get the error:

matmul_firrtl_2.mlir:5:65: error: 'rtl.instance' op requires attribute 'instanceName'

I see @mikeurbach encountered a similar issue in #463 where he closed the case (based on #407 and his merge in #485) but now I am facing the same issue. 

See attached files for reference.

[MatmulFiles.zip](https://github.com/llvm/circt/files/6039636/MatmulFiles.zip)

youngar, 2021-02-25:
Hi!  Thanks for reporting this, I just starting looking into it.

1. In #463 it looks like we never started supporting instance with no names
2. I can modify the memory blackboxing to add instance names based on the memory name, circumventing 1
3. After that, there is a problem with blackboxing write memories

I implemented 2 and 3 locally, which I will open PRs for.  In the mean time, I've attached the final verilog.

[new_matmul.v.zip](https://github.com/llvm/circt/files/6040752/new_matmul.v.zip)

mikeurbach, 2021-02-25:
> Despite what is said in #463, it looks like we never started supporting instances with no names

I think it is important to differentiate between no `instanceName` attribute, and an `instanceName` attribute set to `""`. I believe the former still leads to this error, but the latter should lead to an auto-generated name for the instance in Verilog output.

The change looks good to me in #674, but for the record, what I mean is if this [line](https://github.com/llvm/circt/pull/674/files#diff-41d0d6c5c4f04cfd65eaf651d71ce8e10d12c667cc1ce7e543a220bed18d78f2L133) was a StringAttr with an empty string instead of a null StringAttr, that should have worked.

But implementing 2. sounds like what we really want, thanks @youngar.

JuanEsco063, 2021-02-25:
#674  is merged, solving this issue.

closed 2021-02-25 by JuanEsco063

youngar, 2021-02-25:
@JuanEsco063  Small heads up that the "emit-wrapper=false" option for blackboxing memories is not able to lower all the way to verilog yet, we need to support wires with bundle types in firrtl-lower-types.

> StringAttr with an empty string instead of a null StringAttr, that should have worked.

That is helpful, I didn't realize this. I saw that the `instanceName` was Optional<> in FIRRTL, but not in RTL. Would it make sense to make this attribute optional in RTL, or at handle this difference in [LowerToRTL](https://github.com/llvm/circt/blob/f8ac1985c993cae6e8da73398d28edb0fe1e9c19/lib/Conversion/FIRRTLToRTL/LowerToRTL.cpp#L714-L717)?

mikeurbach, 2021-02-25:
> Would it make sense to make this attribute optional in RTL, or at handle this difference in LowerToRTL?

I think that was discussed in https://github.com/llvm/circt/issues/463#issuecomment-761236603. IMHO that section of `LowerToRTL` you linked to could potentially have an `else` branch that created an `instanceName` attr with `""` for the value to appease the verifier.

>  we need to support wires with bundle types in firrtl-lower-types.

Do you mean memories with bundle typed values? Most of what firrtl-lower-types does is lower wires with bundle types, and that already works for the Handshake use-case, which doesn't ever put bundles into memories.

youngar, 2021-02-25:
I meant wires with bundle types, I didn't see support - maybe I'm missing the place?  I started adding support [here](https://github.com/llvm/circt/commit/67bf4f1eb7c99495c1b65e82a2a8b2c6da116ab4), but I think its lowering the connect statement wrong when there is a duplex type.

mikeurbach, 2021-02-25:
Sorry, I saw #680 and now I understand.