llvm/circt #671: [FIRRTL] (rem / mux) missmatch

drom, 2021-02-25:
The following FIRRTL program

```python
circuit top_mod :
  module top_mod :
    input tmp13: SInt<11>
    output tmp28: UInt<26>
    input _tmp24: UInt<1>
    input tmp25: UInt<26>
    wire tmp18: SInt<11>
    tmp18 <= rem(tmp13, asSInt(UInt<24>(9972122)))
    tmp28 <= mux(tail(asUInt(tmp18), 10), tmp25, _tmp24)
```

Compiled with `firtool --lower-to-rtl --enable-lower-types` produces this Verilog:

```verilog
// skiped defines
module top_mod(
  input  [10:0] tmp13,
  input         _tmp24,
  input  [25:0] tmp25,
  output [25:0] tmp28);

  wire [10:0] tmp18;	// a_top_mod.fir:7:5

  wire [23:0] _T = $signed({{13{tmp13[10]}}, tmp13}) % -24'sh67D666;	// a_top_mod.fir:8:{14,25}
  assign tmp18 = _T[10:0];	// a_top_mod.fir:8:{11,14}
  wire [10:0] _T_0 = tmp18;	// a_top_mod.fir:9:18
  assign tmp28 = _T_0[0] ? tmp25 : {25'h0, _tmp24};	// a_top_mod.fir:2:3, :9:{14,18}
endmodule
```

Compiled with `firrtl-1.5-SNAPSHOT --dont-fold div` produces this Verilog:

```verilog
module top_mod(
  input  [10:0] tmp13,
  output [25:0] tmp28,
  input         _tmp24,
  input  [25:0] tmp25
);
  wire [23:0] _GEN_0 = {{13{tmp13[10]}},tmp13};
  wire [23:0] _GEN_1 = $signed(_GEN_0) % -24'sh67d666;
  assign tmp28 = _GEN_1[13:0] ? tmp25 : {{25'd0}, _tmp24};
endmodule
```

`Yosys 0.9+3911 (git sha1 4b31223e, clang 11.0.1 -fPIC -Os)` reports formal mismatch:

```
ERROR: Found 26 unproven $equiv cells in 'equiv_status -assert'.
```

lattner, 2021-02-28:
I'm pretty sure this is an SFC bug, note the last line:

`assign tmp28 = _GEN_1[13:0] ? tmp25 : {{25'd0}, _tmp24};`

It should only be looking at the low 11 bits of the remainder result, not the low 14 bits, because the condition to the mux is `tail(asUInt(tmp18), 10)`.

@seldridge do you agree?

seldridge, 2021-02-28:
tl;dr: Looks like an SFC bug. Emission of a mux with a multi-bit condition is clearly wrong here.

Couple of things:

1. The SFC is allowing muxes (or whens) to use a selection signal that isn't one-bit. E.g., it accepts the following while CIRCT (rightly?) rejects this:

```scala
circuit Foo:
  module Foo:
    input sel: UInt<10>
    input a: UInt<1>
    input b: UInt<1>
    output c: UInt<1>

    c <= mux(sel, a, b)
```

I opened https://github.com/chipsalliance/firrtl/issues/2093 to track this. This is largely orthogonal to this issue, but might help catch this in the SFC.

2. The issue reported here is similar to https://github.com/chipsalliance/firrtl/issues/2037, but the candidate fix in https://github.com/chipsalliance/firrtl/pull/2038 doesn't fix this. I expect the problem is either in the `InlineCasts` transform or with the interaction of that with how `rem` is cleaned up. Disabling cast inlining into primitive ops globally gets the Verilog to make more sense (the condition is single-bit). However, the FIRRTL IR that's being fed into the emitter looks fine. I think the emitter is just getting this wrong.

lattner, 2021-03-01:
Thank you for confirming this @seldridge !

closed 2021-03-01 by lattner