llvm/circt #672: [FIRRTL] (rem / eq) mismatch

drom, 2021-02-25:
The following FIRRTL program

```python
circuit top_mod :
  module top_mod :
    input tmp22: UInt<5>
    output _tmp23: UInt<1>
    wire tmp16: SInt<2>
    tmp16 <= rem(SInt<2>("o-1"), SInt<14>("o16271"))
    _tmp23 <= eq(tmp22, asUInt(tmp16))
```

Compiled with `firtool --lower-to-rtl --enable-lower-types` produces this Verilog:

```verilog
// skiped defines
module top_mod(
  input  [4:0] tmp22,
  output       _tmp23);

  wire [1:0] tmp16;	// a_top_mod.fir:6:5

  wire [13:0] _T = -14'sh1 % 14'sh1CB9;	// a_top_mod.fir:7:{14,34}
  assign tmp16 = _T[1:0];	// a_top_mod.fir:7:{11,14}
  assign _tmp23 = tmp22 == {3'h0, tmp16};	// a_top_mod.fir:3:3, :8:15
endmodule
```

Compiled with `firrtl-1.5-SNAPSHOT --dont-fold div` produces this Verilog:

```verilog
module top_mod(
  input  [4:0] tmp22,
  output       _tmp23
);
  wire [13:0] _GEN_0 = -14'sh1 % 14'sh1cb9;
  wire [13:0] _GEN_1 = _GEN_0;
  assign _tmp23 = tmp22 == _GEN_1;
endmodule
```

`Yosys 0.9+3911 (git sha1 4b31223e, clang 11.0.1 -fPIC -Os)` reports formal mismatch:

```
ERROR: Found 1 unproven $equiv cells in 'equiv_status -assert'.
```

lattner, 2021-02-28:
This looks like an SFC bug, it incorrectly eliminated a truncation to 2 bits due to tmp16.  Do you agree @seldridge ?

seldridge, 2021-02-28:
Yeah, SFC bug. Superficially, this may be the same thing as #671. The issue appears to be related to use of `asUInt(rem(...))` for a `rem` on `SInt` types resulting in (somehow) the bit extract getting removed.

closed 2021-03-01 by lattner