llvm/circt #678: [FIRRTL] (div / reg / add) mismatch

drom, 2021-02-25:
The following FIRRTL program

```python
; seed: 3727
circuit top_mod :
  module top_mod :
    input clock: Clock
    input reset: UInt<1>
    output tmp22: SInt<8>
    output _tmp30: UInt<1>
    input tmp6: SInt<1>
    wire tmp7: UInt<1>
    reg _tmp17: UInt<7>, clock
    reg tmp21: UInt<1>, clock with: (reset => (reset, UInt<1>(0)))
    tmp7 <= UInt<1>(0)
    _tmp17 <= div(UInt<7>("b1101111"), tmp7)
    tmp21 <= orr(_tmp17)
    tmp22 <= add(tmp6, asSInt(_tmp17))
    _tmp30 <= tmp21
```

Compiled with `firtool --lower-to-rtl --enable-lower-types` produces this Verilog:

```verilog
// skiped defines
module top_mod(
  input        clock, reset, tmp6,
  output [7:0] tmp22,
  output       _tmp30);

  wire       tmp7;	// a_top_mod.fir:9:5
  reg  [6:0] _tmp17;	// a_top_mod.fir:10:5
  reg        tmp21;	// a_top_mod.fir:11:5

// skiped ifdefs
  assign tmp7 = 1'h0;	// a_top_mod.fir:11:55, :12:10
  always_ff @(posedge clock)	// a_top_mod.fir:13:12
    _tmp17 <= 7'h6F / {6'h0, tmp7};	// a_top_mod.fir:13:{12,15,19}
  wire [6:0] _T = _tmp17;	// a_top_mod.fir:14:14
  always_ff @(posedge clock) begin	// a_top_mod.fir:14:11
    if (reset)	// a_top_mod.fir:14:11
      tmp21 <= 1'h0;	// a_top_mod.fir:11:{5,55}
    else	// a_top_mod.fir:14:11
      tmp21 <= |_T;	// a_top_mod.fir:14:{11,14}
  end // always_ff @(posedge)
  assign tmp22 = {8{tmp6}} + {{1{_T[6]}}, _T};	// a_top_mod.fir:3:3, :15:14
  assign _tmp30 = tmp21;	// a_top_mod.fir:3:3
endmodule
```

Compiled with `firrtl-1.5-SNAPSHOT --dont-fold div` produces this Verilog:

```verilog
module top_mod(
  input        clock,
  input        reset,
  output [7:0] tmp22,
  output       _tmp30,
  input        tmp6
);
// skiped ifdefs
  reg [6:0] _tmp17;
  reg  tmp21;
  wire [6:0] _GEN_0 = {7{tmp6}};
  assign tmp22 = $signed(_GEN_0) + $signed(_tmp17);
  assign _tmp30 = tmp21;
  always @(posedge clock) begin
    _tmp17 <= 7'h6f / 1'h0;
    if (reset) begin
      tmp21 <= 1'h0;
    end else begin
      tmp21 <= |_tmp17;
    end
  end
// skiped ifdefs
endmodule
```

`Yosys 0.9+3911 (git sha1 4b31223e, clang 11.0.1 -fPIC -Os)` reports formal mismatch:

```
yosys 0
ERROR: Found 9 unproven $equiv cells in 'equiv_status -assert'.
yosys 1
ERROR: Found 1 unproven $equiv cells in 'equiv_status -assert'.
```

lattner, 2021-02-28:
I didn't analyze this deeply, but I think this is just yosys getting confused by the division by zero.  Does this still show a difference if you remove that?

seldridge, 2021-02-28:
This may be subtly different. I don't know how Yosys is viewing this... The division looks, superficially, to both be dividing by zero, but requires Yosys to understand that `7'b0` and `1'b0` are the same and that the constant propagation is okay. I don't know if it views the constant propagation as making these equivalent or not.

I'd be curious if manually doing the constant prop makes these equivalent (though, I'm biased as I've been thinking we need some good copy propagation/constant propagation inside CIRCT).

lattner, 2021-03-01:
Yeah, constant propagation isn't hard, it is "don't blow away all the wires" that is more tricky.  Once annotations land we will know what wires need to be kept around.

Does SFC constant prop through all wires, even ones with don't touch annotations on them?

seldridge, 2021-03-01:
> Does SFC constant prop through all wires, even ones with don't touch annotations on them?

Yes, if something is marked "don't touch" then it just won't be removed. However, it isn't blocking constant propagation.

It's been discussed that this is too coarse and you really want to have behaviors separately available: (1) something should show up in the output and (2) something should be an optimization barrier.

lattner, 2021-03-02:
Re: wire constprop, Ok thanks!

I don't see an equivalence problem (other than divide by zero) in the above, does anyone else?  If not, I think we should close this as "behaves correctly".

drom, 2021-03-02:
I am not sure that:

```verilog
assign tmp22 = {8{tmp6}} + {{1{_T[6]}}, _T};
```

is equivalent to:

```verilog
wire [6:0] _GEN_0 = {7{tmp6}};
assign tmp22 = $signed(_GEN_0) + $signed(_tmp17);
```

assuming `tmp6` is signed.

lattner, 2021-03-04:
K.  Is anyone able to identify a definitive problem with our output? :-)

seldridge, 2021-03-04:
@drom talked to me about this. The issue appears to be in this sequence:

```verilog
reg  [6:0] _tmp17;
always_ff @(posedge clock)
  _tmp17 <= 7'h6F / {6'h0, tmp7};
wire [6:0] _T = _tmp17;
assign tmp22 = {8{tmp6}} + {{1{_T[6]}}, _T};
```

This appears to be doing a sign extension of an unsigned type (`_T`) by repeating the most-significant bit. That is correct for a signed type, but incorrect for unsigned. So, there may be a problem here with an incorrect signed type getting generated somewhere. I expect that looking at the intermediate pass output would help with this.

darthscsi, 2021-03-05:
Perhaps a quick simulation to see divergence?

lattner, 2021-03-08:
Yeah, I'm happy to help fix this ,but I'd apprecaite help with the analysis.  Something like "this short fir snippet turns into this verilog code, which is incorrect because of X".

lattner, 2021-03-09:
I don't see anything wrong with this.  I think this is completely tied up with the equivalence checker getting confused by division by zero.

closed 2021-03-09 by lattner