llvm/circt #679: [FIRRTL] (cat / gt) mismatch

drom, 2021-02-25:
The following FIRRTL program

```python
; seed: 4046
circuit top_mod :
  module top_mod :
    input clock: Clock
    input reset: UInt<1>
    input inp_h: SInt<0>
    input inp_e: UInt<11>
    output tmp15: UInt<1>
    reg tmp6: UInt<28>, clock
    reg _tmp10: UInt<1>, clock with: (reset => (reset, UInt<1>(0)))
    tmp6 <= cat(SInt<28>("h-6de5f19"), inp_h)
    _tmp10 <= gt(UInt<11>(0), tmp6)
    tmp15 <= _tmp10
```

Compiled with `firtool --lower-to-rtl --enable-lower-types` produces this Verilog:

```verilog
// skipped defines
module top_mod(
  input         clock, reset,
  input  [10:0] inp_e,
  output        tmp15);

  reg [27:0] tmp6;	// a_top_mod.fir:9:5
  reg        _tmp10;	// a_top_mod.fir:10:5
// skipped ifdefs
  always_ff @(posedge clock)	// a_top_mod.fir:11:10
    tmp6 <= 28'h921A0E7;	// a_top_mod.fir:11:{10,17}
  always_ff @(posedge clock) begin	// a_top_mod.fir:12:12
    if (reset)	// a_top_mod.fir:12:12
      _tmp10 <= 1'h0;	// a_top_mod.fir:10:5, :12:15
    else	// a_top_mod.fir:12:12
      _tmp10 <= 1'h0;	// a_top_mod.fir:12:{12,15}
  end // always_ff @(posedge)
  assign tmp15 = _tmp10;	// a_top_mod.fir:3:3
endmodule
```

Compiled with `firrtl-1.5-SNAPSHOT --dont-fold div` produces this Verilog:

```verilog
module top_mod(
  input         clock,
  input         reset,
  input  [10:0] inp_e,
  output        tmp15
);
// skipped ifdefs
  reg  _tmp10;
  assign tmp15 = _tmp10;
  always @(posedge clock) begin
    if (reset) begin
      _tmp10 <= 1'h0;
    end else begin
      _tmp10 <= 1'h1;
    end
  end
// skipped ifdefs
endmodule
```

`Yosys 0.9+3911 (git sha1 4b31223e, clang 11.0.1 -fPIC -Os)` reports formal mismatch:

```
yosys 0
ERROR: Found 1 unproven $equiv cells in 'equiv_status -assert'.
yosys 1
ERROR: Found 1 unproven $equiv cells in 'equiv_status -assert'.
yosys 2
ERROR: Found 1 unproven $equiv cells in 'equiv_status -assert'.
```

lattner, 2021-02-28:
This is an SFC bug, we are correctly folding `gt(UInt<11>(0), tmp6)` aka `0 >u tmp6` to false, it is incorrectly folding it to true.  @seldridge do you concur?

seldridge, 2021-02-28:
Looks like an SFC bug related to zero-width handling.

The SFC is converting:

```scala
tmp6 <= cat(SInt<28>("h-6de5f19"), inp_h)
```

Into:

```scala
tmp6 <= SInt<28>("h-6de5f19")
```

During the pass `firrtl.passes.ZeroWidth`. This looks wrong as it's an illegal type change of `tmp6` from `UInt` to `SInt` which cause the `gt` constant prop. to go the other way.

lattner, 2021-03-01:
Thanks, I'm going to close this from the circt tracker then.

closed 2021-03-01 by lattner