llvm/circt #681: module names with keywords aren't renamed in export

darthscsi, 2021-02-25:
for example, rtl-dialect.mlir has a module "inout" which isn't allowed.

seldridge, 2021-02-27:
Full example:

```scala
circuit inout:
  module inout:
    skip
```

This currently produces:

```verilog
module inout();
endmodule
```

Which Verilator linting catches:
```bash
firtool inout.fir -enable-lower-types -lower-to-rtl -verilog | verilator -lint-only /dev/stdin
# %Error: /dev/stdin:55:8: syntax error, unexpected inout, expecting IDENTIFIER or PACKAGE-IDENTIFIER or TYPE-IDENTIFIER
#    55 | module inout();
#       |        ^~~~~
# %Error: Exiting due to 1 error(s)
```

prithayan, 2021-03-08:
Fixed in https://github.com/llvm/circt/pull/697

closed 2021-03-08 by prithayan